Xinyuan-LilyGO/LilyGo-EPD47 #2: Hardware details

loboris, 2020-12-04:
Which display is used on this product? Could you give us the exact part number?
Is the schematic available somewhere?

lewisxhe, 2020-12-04:
Hi, loboris

I am glad that this can attract your attention. We are using the ink screen of VINTEK from Taiwan. I cannot provide the model for the time being. It is not here. The schematic diagram is not available for the time being. If you are interested in helping us, then I will email and You contact and provide you with free samples to help us improve.

loboris, 2020-12-04:
Hi lewisxhe,
thanks for the reply.
I'm quite interested in using you product with ESP-IDF and my port of MicroPython for ESP32.
I have a lot of experience in working with ESP32 and I've just started a project to use a similar display (ED060SC4 6" e-ink) with K210.
If you can provide any additional information please send it to loboris@gmail.com.

BTW, is the display connector pinout on your board compatible with ED060SC4?

lewisxhe, 2020-12-04:
@loboris  
That's great, I knew you a long time ago, but you didn't know me! I know you are a very good person, and it is great that you can help us.

By the way, you can refer to the schematic here https://github.com/vroland/epdiy/tree/master/hardware
Basically the same!

We will contact you via email soon! Thanks again

xeijin, 2020-12-21:
given that the LilyGo-EPD47 is an ESP32, e-ink screen, battery and (optional) touch screen interface, I think it would be perfect for this ESP firmware project:

https://github.com/martinberlin/cale-idf

The [main project, cale.es](https://cale.es/) is an API mash-up service for WiFi displays, [and many eink displays](https://cale.es/eink-displays) are already supported. Think it could be the perfect match.

@lewisxhe maybe you could send @martinberlin (creator of the firmware), a sample too?

[image]

martinberlin, 2020-12-21:
Would like to add support to this, the only thing is that I would have to create a new IO class, since if I see that correctly this Epaper uses parallel protocol instead of SPI. That should be faster but I must say that I don't have experience with I2S anyways looks like a nice challenge. So accepted if I can get one in my hands I will make the ESP-IDF C++ class with GFX support.

lewisxhe, 2020-12-22:
Hi, @martinberlin

I know you, if you want, we can send you an e-paper and let you help us improve the firmware! Please send your address to jiangcuifeng@lilygo.cc this mailbox!

martinberlin, 2020-12-22:
Done Lewis. Many thanks for the opportunity to try this out!

markbirss, 2020-12-23:
> Hi, @martinberlin
> 
> I know you, if you want, we can send you an e-paper and let you help us improve the firmware! Please send your address to [sales@lilygo.cc](mailto:sales@lilygo.cc) this mailbox!

Im also interested in this product as I want to use it for a terminal display

fdufnews, 2021-01-04:
Can you please publish the board's schematic.
It will help using the board. For example, the board is supposed to be used in low power applications and there is a blue LED that is all the time on. Is this LED driven by ESP32 or directly connected to the power line? I would like to switch it off to reduce consumption.

vroland, 2021-01-05:
@martinberlin Since your firmware is written as an IDF component, you could also use the epdiy library directly: https://github.com/vroland/epdiy. This board is based on epdiy and I'll soon add support for it in the driver. That way, you don't need to depend on Arduino and add support for other parallel e-Paper displays as well.

fpasteau, 2021-01-06:
I would also need the schematics. In a low power application with battery, we need to know how much the battery is depeted to avoid running it too low. Right now, I don't know if the battery voltage has been wired to any of the ESP32 ADC pins for us to read.
I would also like to turn off this blue led as it uses unnecessary power.

martinberlin, 2021-01-08:
Great tip @vroland I'm bookmarking this repository. Waiting anxiously that this new display comes.
When I find some time will design also a 3D model for the [Housing](https://www.thingiverse.com/movil/designs).

markbirss, 2021-01-08:
@Martin not sure if you saw this already in the repository ?
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/3D_file

martinberlin, 2021-01-08:
Ahem! Nice. But what kind of 3D format is that?
I cannot import it in Blender. Also not in Slicer, so I cannot really printed until is transformed in some other format

ElementW, 2021-01-09:
On the touchscreen sold with the EPD47: the store page mentions "_Touch chip: L58_", but this yields no relevant search results.
The screen is the same as a Samsung Galaxy S III, which uses a Melfas 8PL533 controller ([iFixit](https://www.ifixit.com/Teardown/Samsung+Galaxy+S+III+Teardown/9391), [Linux defconfig](https://github.com/pascua28/Elite-Boeffla-Kernel/blob/4681db377b36df8be68dada7786be543878a6f43/arch/arm/configs/i9300_defconfig#L1665)), but the package, pinout, and I2C comm is different.
Unfortunately the IC is unmarked; I've been looking for what it could be but no dice.
@lewisxhe Do you know the name of the touchscreen controller IC?

kd8bxp, 2021-01-10:
@martinberlin No expert at cad or 3d design, the closest I get is tinkercad - I've made a few things, but just FIY
I used freecad to open the stp file, and after some time hitting buttons :-)  was able to convert the main part of the case to an stl file, haven't printed it yet - so don't know how well it will work, but it looks good.  
The buttons look like they will print really small, and the top I think I messed up, it doesn't have a hole for the display.
I'm also no expert on any cad software, so it is entirely possible I did something wrong.
I'll print the case tomorrow and at least see how that works.

markbirss, 2021-01-10:
@martinberlin 
Hope it helps, a online converter was used
[0-sjk_asm(2).zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5792586/0-sjk_asm.2.zip)

markbirss, 2021-01-10:
@ElementW 

not sure what is actually used but here with the LiLyGo EPD47

The M5Paper uses the GT911 for touch controller
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/m5paper/gt911_datasheet.pdf

martinberlin, 2021-01-10:
I added a new issue #10 about this 3D model issue so we don't spam @loboris with things that have nothing to do with the issue he posted. I also think that a proper schematics will be great so we can unleash the full potential of this. 
I have the same issues with other TTGO products like the small T5 epapers. Which are great and I really like to do stuff on them but they lack a bit proper schematics & documentation. Let's give them some time so they can do it properly this time for this parallel epaper that looks like a great product.

lewisxhe, 2021-01-11:
> On the touchscreen sold with the EPD47: the store page mentions "_Touch chip: L58_", but this yields no relevant search results.
> The screen is the same as a Samsung Galaxy S III, which uses a Melfas 8PL533 controller ([iFixit](https://www.ifixit.com/Teardown/Samsung+Galaxy+S+III+Teardown/9391), [Linux defconfig](https://github.com/pascua28/Elite-Boeffla-Kernel/blob/4681db377b36df8be68dada7786be543878a6f43/arch/arm/configs/i9300_defconfig#L1665)), but the package, pinout, and I2C comm of the sold screen is different.
> Unfortunately the IC is unmarked; I've been looking for what it could be but no dice.
> @lewisxhe Do you know the name of the touchscreen controller IC?

Unfortunately, we don’t have detailed information about this touch screen, but only an introduction to it. The sample code in the warehouse is still modified by the code provided by the manufacturer. For multi-touch, it still has problems, but for single-touch , It works normally.

lewisxhe, 2021-01-11:
This issueis too redundant, please open in a new issueis

closed 2021-01-11 by lewisxhe

markbirss, 2021-01-11:
> Can you please publish the board's schematic.
> It will help using the board. For example, the board is supposed to be used in low power applications and there is a blue LED that is all the time on. Is this LED driven by ESP32 or directly connected to the power line? I would like to switch it off to reduce consumption.
> 
> Edit: I noticed that the blue LED is switched off when the ESP32 goes into deep sleep. So I suppose the LED is driven by the ESP. Nevertheless still interested by the board's schematic

MrNiceThings made a board based on this design also

https://www.reddit.com/r/esp32/comments/k4uxvh/6_kindle_paperwhite_display_with_esp32/

[Schematic_9.7inch e-ink controller_2020-12-06_20-17-59.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5795181/Schematic_9.7inch.e-ink.controller_2020-12-06_20-17-59.pdf)

vke, 2021-01-11:
[schematic.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5795308/schematic.pdf)

fdufnews, 2021-01-11:
@lewisxhe if it is the official schematic it would be cool to store it on the repository as there is little chance people will go in the closed issues to search for the schematic.

vke, 2021-01-11:
They must export the 3D PCB model from Altium to a STEP file also for users who wish to draw their own enclosures.

fpasteau, 2021-01-11:
> Can you please publish the board's schematic.
> It will help using the board. For example, the board is supposed to be used in low power applications and there is a blue LED that is all the time on. [...]

From the schematics, this blue led seems to be LED7, directly connected to the 3V3 power rail. This 3V3 power rail is the one powering the screen and the P4/P5 4 pin plugs, it differs from the VDD3V3 power rail powering the ESP32. You can turn this led on/off using the epd_poweron() and epd_poweroff() function but it will also power on/off the screen. If you want to completely turn it off, you may have to remove it or its resistor.

As for sensing the battery level, it can be done using ADC on GPIO36 which is connected to the battery with a voltage divider of 1/2 ratio.

On a side note, this schematics also explain a strange screen behavior that I had. When unplugging the usb cable after displaying some image on the screen, the display started to fade out a little bit, seemingly losing contrast. As a matter of fact, the 3V3 power rail is always turned on when power is applied through USB. Therefore when disconnecting the USB cable, the 3V3 power rail is turned off but without following the timing the screen expects leading to a this strange behavior. Furthermore, it also means the if you're code is running well when USB is plugged, it does not mean it will run the same way when running on batteries ( you may forget to turn on/off the screen as it is always on when USB is connected).

markbirss, 2021-01-11:
@fpasteau as this design was based on https://github.com/vroland/epdiy/ these issues will exist whilst epdiy is undergoing a further revision to address some power issues

https://github.com/vroland/epdiy/issues/50
https://github.com/vroland/epdiy/issues/47

Kabron287, 2021-01-20:
I noticed that dual DC/DC converter U2 could not be shutdown.
Are there free MCU pins to route to SHDN1/SHDN2 pins of U2?

markbirss, 2021-02-04:
Here my LilyGO 4.7 e-Paper Raspberry Pi Zero W Terminal with USB keyboard
95 x 25 characters

[image]

Thanks for your help from @vroland

krystiansliwa, 2021-02-04:
> Here my LilyGO 4.7 e-Paper Raspberry Pi Zero W Terminal with USB keyboard
> 95 x 25 characters
> 
> [image]
> 
> Thanks for your help from @vroland

wow amazing, how you connect board to the RPi?

markbirss, 2021-02-04:
RxD TxD GND

G6EJD, 2021-02-04:
It's a serial port connection, so displaying all serial port data.

markbirss, 2021-02-04:
Yes with some modifications to Pi like what you would use your kindle touch
and pi is the past

vroland, 2021-02-04:
I'd recommend adding flow rts/cts flow control if you use more complex things like ncurses GUIs. With things like nmtui or even raspi-config (or console media players), data can get lost otherwise, corrupting the output.

markbirss, 2021-02-04:
yes, still todo waiting to see what GPIO's are left as they are not printed correctly on the silk screen

G6EJD, 2021-02-04:
Which’s pins have you defined and used for the serial connection? 12,13 or 14, 15? That will help confirm which pins are in use v the silkscreen, thanks

markbirss, 2021-02-04:
@G6EJD 

At the moment im using TxD - 13 and RxD 12

G6EJD, 2021-02-04:
Thanks, I’m waiting for a molex connector to try out mine. GPIO-12 is a little tricky as if held high during boot it changes the ESP32 configuration.

markbirss, 2021-02-04:
@G6EJD 

I chopped a M5 Stack cable (made it round)  and soldered a little break-out, and yes GPIO-12 is tricky
https://m5stack.com/products/4pin-buckled-grove-cable

markbirss, 2021-02-05:
[image]

mengszym, 2021-02-16:
Hello, after uploading the water program to epd47 after refreshing the screen, the gray background remains.  how to fix it ?  Thank you in advance for your help and best regards

fdufnews, 2021-02-16:
What do you mean with "water progam"?

mengszym, 2021-02-16:
Hello, I mean the program that is shown in the pictures on the shares with
the device, I got it from you in the EPD47 file in the sample folder there
was a program called wather and after uploading this program there is a
gray background and the weather icons and descriptions are hard to see.
grayscale and picture work fine on demo programs.
  EPD47.zip
<https://drive.google.com/file/d/172DdLXkLKNmPq4w0pVGuinxBqQSDX6fN/view?usp=drivesdk>

mengszym, 2021-02-16:
wt., 16 lut 2021, 11:38 użytkownik timon w <sornydror@gmail.com> napisał:

fdufnews, 2021-02-16:
You are not running the right software. The one you run is for another configuration.
you shall use this one [https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display](url)

mengszym, 2021-02-16:
unfortunately the link does not work: /

G6EJD, 2021-02-16:
I’ve just checked if and it’s OK?

mengszym, 2021-02-16:
now it works great :) :) :) thank you very much you are a master greetings
:) you are great :)

mengszym, 2021-02-20:
hello, is there a weather station program for LilyGo 1.54 "epaper or some other interesting program?[image]