Xinyuan-LilyGO/LilyGo-EPD47 #4: License

vroland, 2020-12-12:
Hi, i have mentioned this in https://github.com/vroland/epdiy/issues/44, but just as a reminder:

Please note that the driver and examples (at least the ones written by me) are licensed under the GPLv3 (https://github.com/vroland/epdiy/blob/master/components/epd_driver/LICENSE), which requires that derivatives must be licensed under the same terms.
Currently, this Repo does not contain any licensing information, so please choose a license that is in accordance with the original license.

Regards,
Valentin

lewisxhe, 2020-12-12:
Sorry, but forgot to add a license. It has been added, thank you for your contribution! 😂

vroland, 2020-12-12:
Thank you! :+1:

closed 2020-12-12 by vroland