Xinyuan-LilyGO/LilyGo-EPD47 #6: Core 1 Panic: (StoreProhibited). Exception was unhandled.

hpsaturn, 2021-01-03:
Hi,

I have a new device T5-4.7, I tried to build the demo example but I got:

```bash
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c46c  PS      : 0x00060130  A0      : 0x800d0f10  A1      : 0x3ffb1f70  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0003f480  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00003f48  A8      : 0x00000000  A9      : 0x3ffb1f40  
A10     : 0x4008f954  A11     : 0x000001e0  A12     : 0x00000000  A13     : 0x3ffd1cfc  
A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00003f47  

Backtrace: 0x4000c46c:0x3ffb1f70 0x400d0f0d:0x3ffb1f80 0x400d4cfb:0x3ffb1fb0 0x40089195:0x3ffb1fd0
```

I have the next parameters with PlatformIO (similar to Arduino instructions that you posted here):

```python
platformio]
src_dir = .

[common_env_data]
framework = arduino
upload_speed = 1500000
monitor_speed = 115200
lib_deps =
    Wire
    https://github.com/Xinyuan-LilyGO/LilyGo-EPD47.git
build_flags =
    -DBOARD_HAS_PSRAM

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = ${common_env_data.framework}
upload_speed = ${common_env_data.upload_speed}
monitor_speed = ${common_env_data.monitor_speed}
lib_deps = ${common_env_data.lib_deps}
```

The build and upload pass ok, but I have this.

Thanks.

lewisxhe, 2021-01-03:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
board_build.f_cpu = 80000000L
framework = arduino
monitor_speed = 115200

build_flags = 
    -DBOARD_HAS_PSRAM
```

Everything is normal for me.
[image]

hpsaturn, 2021-01-03:
Thanks for your answer, I added the cpu frequency line but I had the same core panic, but I resolved the issue, I missing extended the build_flags to my extended environment. The final config for me was:

```
[platformio]
src_dir = .

[common_env_data]
framework = arduino
board_build.f_cpu = 80000000L
upload_speed = 921600
monitor_speed = 115200
lib_deps =
    Wire
    https://github.com/Xinyuan-LilyGO/LilyGo-EPD47.git
build_flags =
    -DBOARD_HAS_PSRAM

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = ${common_env_data.framework}
upload_speed = ${common_env_data.upload_speed}
monitor_speed = ${common_env_data.monitor_speed}
lib_deps = ${common_env_data.lib_deps}
build_flags = ${common_env_data.build_flags}
```

closed 2021-01-03 by hpsaturn