Xinyuan-LilyGO/LilyGo-EPD47 #8: Tutorial on Fonts and Images

kd8bxp, 2021-01-08:
I've tried to create a custom font, with no luck. I've used the fontconvert.py tool and continue to get and error "valueError: code point 166 not found in font stack!".
I've tried to convert a ttf, and a otf. With no luck.
Is there a tutorial somewhere on how to create a font for the T5 4.7" 

a tutorial on creating a custom image would also be helpful, thank you.

kd8bxp, 2021-01-09:
Update: I was able to get an image to convert and work, but still having issues with custom fonts. 
thanks

lewisxhe, 2021-01-09:
Prerequisite you need to install python3 and install `freetype-py` using pip
The approximate process is like this:

```
sudo apt install python3-pip
python3 -m pip install freetype-py
``
The previous is the prerequisite for implementation, and then you need to store the font file you want to convert in the same path as fontconvert.py. This is just for more convenient operation. You can also fill in the font path.

Then just follow the command below to convert the font.
```
python3 fontconvert.py --compress demo 16 msyh.ttc> font.h
```

Explanation of specific parameters:

```
python3 fontconvert.py --compress [generated font name] [font size] [font file path]> [generated font file]
```

Of course, this only demonstrates the generation of standard ascii codes. If you need other fonts, you only need to fill in the unicode encoding of the font to be generated in the `fontconvert.py`  `intervals` list.

![demo](https://user-images.githubusercontent.com/22990954/104092951-446c1880-52c2-11eb-852e-8c50626eb340.png)

lewisxhe, 2021-01-09:
By the way, please make sure that the unicode encoding in the `fontconvert.py` `intervals` list is included in your font file, otherwise please comment other encodings and only keep the 32,126 range!

[image]

kd8bxp, 2021-01-09:
Thank you, 
Changing the intervals is what the problem was.
This is good information for anyone else who.

kd8bxp, 2021-01-15:
Saw this information was added to a README in the scripts directory, I will close this issue. It's good information. Thanks

closed 2021-01-15 by kd8bxp