Xinyuan-LilyGO/LilyGo-EPD47 #10: 3D printing files

martinberlin, 2021-01-10:
Comes from #2 :
Thanks @markbirss the intention  of converting the file to 0-sjk_asm(2).zip was great but it really does not help like this. When importing it to blender or even the Slicer software this comes as a single block piece that looks like a solid mobile device and it's really useless for 3D printing. 
Preview:
[image]

kd8bxp, 2021-01-10:
[T5_EDP47_Case.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5792949/T5_EDP47_Case.zip)
This is mine, as I said in the previous topic, I'm no expert at what I've done. I used FreeCAD and pulled everything apart.
The top seems to be a problem, and maybe I did something wrong.
When you use FreeCAD you can see if  you remove the top, there is a rendering of the T5 display already in the case, there also appears to be a battery (or maybe 2 batteries for the two different models)
The buttons look to me like they will print very small as well, and I'm probably not going to even try to print them.
I am printing the main case now, and can post pictures if it would help.

kd8bxp, 2021-01-10:
Here are pictures from FreeCAD, and I tried again to get the top cover converted to a stl file with no luck.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kd8bxp, 2021-01-10:
here are a few more pictures from FreeCAD
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kd8bxp, 2021-01-10:
Print finished, overall, it turned out good - my printer is a little out of level, so some issues with the bottom (but that is me).
I will try to print the buttons, they will be needed.  It will need a top to hold the board in place, but the biggest problem I see with it right now. The USB-C cable will not fully insert into the provided hole, so, no power from USB-C. Here are the pictures of my print, took almost 3 hours, using a Wanhao Duplicator i3 plus, 5% infill, .15mm - no rafts or supports. 
It does kind of look like a phone case, maybe that is what inspired this design (?)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kd8bxp, 2021-01-10:
Sorry to keep posting like this, but really hoping it helps out in getting a nice case designed for this board. 
I think the buttons need a little work, the holes just seem too small (at least to me) and to me the buttons seem too short. They printed just fine, in about 2 mins time. They were kind of hard to get in place, but I think it is because the holes were just a little small. 
Also I needed to rotate them to be flat on the build plate, I think this could be fixed if I really knew what I was doing when I took apart the larger file. 
[image]
[image]
[image]

lewisxhe, 2021-01-11:
The new 3D files will be available on the 13th, and the ones on github need to be converted to STL before they can be used.

[image]

kd8bxp, 2021-01-11:
@lewisxhe Looking forward to the new case, image looks good.
It looks like you did some work on the USB-C connector.  What about the buttons, are they a little bigger?

martinberlin, 2021-01-11:
Nice, so I have to use Cura then? 
Can it import this **stp** files directly?
I've a Prusa printer and mostly use Slic3r that is a different software to slice 3D Models.

kd8bxp, 2021-01-11:
@martinberlin  no cura uses stl files, I used FreeCAD to convert the STP pull apart the STP file, and convert it the pieces to STL
Cura is just the slicer I use, mainly because it was free and worked with my printer. 
https://www.freecadweb.org/ it's a large download, but it is in the Ubuntu repository if you are using linux, you can get it for Windows and Mac as well.

martinberlin, 2021-01-11:
Can you please upload the STL files here or in Thingiverse so I don't have to install Freecad?
Thanks a lot in advance (Github let's you just drag & drop zip files in Issues)

kd8bxp, 2021-01-11:
already did see above (https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/10#issuecomment-757507333)
it has a problem with the top, and I think the buttons are too small.
Also @lewisxhe said a new 3d file is going to be published on the 13th in a couple of days.

lewisxhe, 2021-01-14:
@martinberlin  @kd8bxp STL file has been uploaded

kd8bxp, 2021-01-14:
@martinberlin  Pulled apart the stp file again, here is what I came up with. hopefully it will help, still looking to get a nice case for this display. 

This is the bottom, and the frame in two separated stl files.
[alternativestls.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5812524/alternativestls.zip)

This is the bottom and the frame as one print 
[caseT5-2.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5812526/caseT5-2.zip)

This is the buttons - they will need to be rotated to fit correctly on the build plate of your printer, there is also a file called "needed?.stl" when I pulled apart the stp file, it looks like some extras maybe to make the buttons longer and fit better - I'm not sure about them. 
[buttons.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5812527/buttons.zip)

It looks like someone will need to design a top, when I pulled apart the stp the top just came out as a whole piece again, without a hole for the display.

Hope this helps, I'm going to print mine in the morning, and can post some photos of it when it is finished.

kd8bxp, 2021-01-14:
Here are some pictures from freeCAD
[image]
[image]
[image]
[image]
[image]

Below is the extra pieces for the button, which I'm not sure if they are needed or not, see the above comment -I suspect they will be needed, but until I get the case printed, it is hard to say.
[image]

martinberlin, 2021-01-14:
Good work, looks much better. Do not forget to activate supports when printing this model!

kd8bxp, 2021-01-15:
Here are my thoughts on the 2nd version of the case.
1) USB-C still not useable
2) Buttons need to be longer (taller), the radius of them appears to be correct. The holes in the frame are too small, the buttons are very very tight and hard to push into the case itself. The hole needs to be slightly larger radius.
3) Still needs a top, After talking a bit to @martinberlin on twitter, we decided the case was designed with the touch screen being used. Need a top for those of us who don't have the touch screen.
4) need a bottom for the case that doesn't have the hole for the 18650 battery (for those of us that are using a lipo) 

I know it's hard to see in these pictures, but I try to show the USB-C cable is still too short to fit, the easy fix for this is to make the hole for the USB-C larger, so that the plastic around the connector can also fit into the case (I took a rotatary tool and made the hole larger, this worked, so probably needs to be in the design).

[image]
[image]

Over all the case printed nicely:
[image]
[image]

Here is the 1st version compared to the 2nd version:
[image]

2nd version with the T5 in place:
[image]
[image]
[image]

very hard to see the button problem - (I think probably just making the holes slightly larger will fix at least one issue) The buttons are flush with the case, and nearly impossible to push (have to really force them to get them into the hole, even after cleaning the hole with a drill bit) it's an very tight fit.
[image]

This is the USB-C cable after I cut the hole bigger:
[image]
[image]
[image]
[image]

kd8bxp, 2021-01-15:
Forgot to put the size of the USB-C hole after I cut it out a bit:

TOP to BOTTOM 7.0mm to 7.5mm (I used a rotary cutting tool, so it's not percsie at all)
SIDE TO SIDE: 12.4mm to 12.6mm
These are all approximate - I would probably error on making the hole bigger then smaller. So 12.6 and 7.5mm 

My USB-C cables is 6.0mm and 10.5mm approximate.  so the above numbers seem to good size for the hole. 
[image]
[image]

martinberlin, 2021-03-01:
Totally agree with the USB-C size. It should fit most common connectors out of the printer, I also had to enlarge mine with a cutter and looks quite bad ;)

martinberlin, 2021-03-02:
**Warning:** If you assemble the touch overlay take great are when connecting the small flat cable. The connector in the Lilygo PCB is not meant to open and close many times. I got one broken just by trying to get it inside the 3D printed housing.

The touch connector needs a small 3D printed part so the cable is not doing an inverted U form to the top. Otherwise is very dangerous to put it in such a small housing because you risk breaking it (Like I did)
Either it needs to be more flat to the PCB (is too large for this)
or this side of the 3D model should give a bit more space to avoid making pressure on the small FPC cable when mounting it inside the housing.

kd8bxp, 2021-03-07:
@martinberlin Nice again, I am printing it now using PLA (only material I have at the moment), thanks for the update

guyou, 2021-03-11:
Please, note that we can find more models on platforms like tingiverse. Examples:
- https://www.thingiverse.com/thing:4782302
- https://www.thingiverse.com/thing:4749075

martinberlin, 2021-03-11:
Second one looks nice. But attention that the buttons seem all in one piece, I guess that is supposed to break and move, but I'm not sure if it will work like expected with all printers. I've printed mine with the original in 2 pieces, the down part closed in PLA, the sides in FLEX. Looks nice, waiting for the new epaper since I give mine to a friend!

kd8bxp, 2021-03-11:
@guyou Thank you for sharing, I think I like the 1st one a bit more - And they look like they solve at least some of the problems with having a bezel for the model without the touch screen.
Again thanks for sharing.

kd8bxp, 2021-03-12:
@martinberlin @guyou  I printed both cases from above, over all I like both, but the 2nd one does seem to work better. (It really depends on what you want to do with your display thou.) 

I am still having some issues with the buttons (on both cases, but over all these are good.)

So This is my print of https://www.thingiverse.com/thing:4782302
[image]
With the buttons in place, the display is held in place - without the buttons, the display will fall. The USB-C connector is on the bottom, there is no room for a 90 degree adapter - the author of the case said they are running on battery. So this is great for that.  
I want to try to make a base that will raise it up so that either a USB-C or a 90 degree adapter will fit under it.

The 2nd case: https://www.thingiverse.com/thing:4749075

[image]
The buttons are better on this one - but very hard to break loose, with the thumb screw in place it works good. And the USB-C connector is easy enough to use. 
The original STL file has the screw holder next to the case, which for me just pushed the print outside of my printer.
I did a small remix where I moved the holder to the inside of the case while it's printing, this saved some room, and only increased the print time by about 4 minutes. I think that is a good trade off. 
My "remix" is here: https://www.thingiverse.com/thing:4791483 
But I really don't like calling this a remix - because it's more of a move. nothing else was changed. 

@guyou thanks again for sharing these cases.

closed 2022-09-02 by liangyingy