Xinyuan-LilyGO/LilyGo-EPD47 #12: How to use IO0 button ?

phantase, 2021-01-11:
My question is not directly related to the screen itself (which works very nicely following the examples) but is related to the hardware of the T5-4.7, and more precisely to the buttons.

There is 5 buttons at the top of the board, one is the RST (the one at left with screen in front of you), then they are named IO0, IO35, IO34, IO39 (as seen in the pictures at the back of the box of the boad, that we have at [this link](https://d3s5r33r268y59.cloudfront.net/datasheets/22266/2020-12-24-02-59-54/QQ%E5%9B%BE%E7%89%8720201210103358.jpg) ).

I have no problem to use the IO35, IO34, and IO39 (the three buttons at the right), but I am currently not able to find which PIN I must use for the first customisable button (the second one from the left).

When we look at the board, we can see this is the only button without resistors attached to him. Is this button working ? The board description says there is 4 custo buttons.

lewisxhe, 2021-01-12:
IO0 has been used and cannot be programmed.

kd8bxp, 2021-01-12:
@lewisxhe I wrote a small "Button" demo - you may use it as you see if you want.
[Button.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/5799648/Button.zip)

phantase, 2021-01-12:
Thanks @lewisxhe for this info, so in fact we have only 3 available buttons (that's a pitty case 4 buttons is very nice for menu navigation, but 3 buttons is also better than no button 👍 ).

Thanks @kd8bxp for this sketch, looks like my own tests (with the same comments added to the definition and use of the first button 😄 

Maybe your sketch can be added to the repo in the examples for other people.

Thanks again, that's good for me.

closed 2021-01-12 by phantase

kd8bxp, 2021-01-12:
@phantase thank you, I'm sure it could be improved upon.  My hope was it might help someone else.