Xinyuan-LilyGO/LilyGo-EPD47 #13: How do you read the battery voltage?

atrueresistance, 2021-01-14:
Looking for a way to get the battery percentage from the board. I looked at the buttons example you posted and I see you defined `#define BATT_PIN            36`. Can this be used to read the battery voltage? 

`float voltage = analogRead(36) / 4096.0 * 7.46;` seems to come out null.  Is there a way without wiring into [additional GPIO](https://randomnerdtutorials.com/power-esp32-esp8266-solar-panels-battery-level-monitoring/#:~:text=One%20way%20to%20do%20that,the%20voltage%20from%20the%20battery.) to check the voltage?

fpasteau, 2021-01-14:
AFAIK according to the schematics you need to have PWR_EN = 3V3 to read the battery voltage. You can do this with the  epd_poweron() function. After doing analogRead, you can power off the screen using epd_poweroff().

lewisxhe, 2021-01-15:
IO36 is configured as voltage detection. It uses a voltage divider of 100k ohms up and down. As @fpasteau said, you need to use edp_poweron() to enable the voltage detection function.

G6EJD, 2021-01-24:
epd_poweron() does nothing, its defined but not implemented.
I see the PWR_EN label on the Schematic, but it is not connected to the ESP32 - that I can see.

The battery voltage reading works without issue.

closed 2021-10-21 by LilyGO