Xinyuan-LilyGO/LilyGo-EPD47 #17: Powersave mode for the touchscreen?

ElementW, 2021-01-19:
At idle, the touchscreen sold with the EPD47 consumes approximately 2 mA. The ESP32, when in deep sleep, consumes power in the range of microamperes; thus making the touchscreen the biggest power consumer on the board.

For wire-powered applications this is acceptable, however when battery-powered from e.g. a 2200 mAh 18650 cell this would drain power in under 1100 h = ~48 days.
Not too short in its own regard, but considering the device will probably be idle most of the time it would be great if the touchscreen power draw could be reduced further.

Removing power from the screen could've been a possibility in certain scenarios but the schematics show it is directly connected to `VDD3V3` which can't be disabled through `PWR_EN` like `3V3` can.

I know the touchscreen controller isn't identified (https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/2#issuecomment-757579958), but is there information on how to trigger some sort of powersave mode, or to reduce the idle screen scanning rate?

lewisxhe, 2021-01-19:
Unfortunately, for the touch controller, we do not have detailed information. Although the manufacturer provides a sleep command, I have tested it and it can reduce about 500uA, the minimum is only 1.5mA, if there is no need to touch, I think it is still disabled to save power consumption.

G6EJD, 2021-01-21:
My initial look at the schematic shows the touch panel is powered from the 3v3 line rather that the switched power for the EPD. 
My advice is get the designs reviewed by someone before committing or note these issues so they can easily be resolved, for what I’ve seen of the current issues they can be resolved with little difficulty and costs should be minimal, it depends who owns the design rights.
I am pleased to help out and try to solve these issues.

closed 2021-10-21 by LilyGO