Xinyuan-LilyGO/LilyGo-EPD47 #18: SCHEMATIC

Kabron287, 2021-01-20:
When you launch new product, the **FIRST** and **MANDATORY** thing that must be done is the 
**SCHEMATIC** of the product! 
That is why you lose the competition with **M5Stack** products.

vke, 2021-01-20:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/2#issuecomment-757875009

Kabron287, 2021-01-20:
Thanks!
I guess it have to be placed in hardware folder of this repo.

vke, 2021-01-20:
Yes, with 3D model of PCB exported from Altium.

Kabron287, 2021-01-20:
I noticed that dual DC/DC converter U2 could not be shutdown.
Are there free MCU pins to route to SHDN1/SHDN2 pins of U2?

fdufnews, 2021-01-20:
I was expecting POS_CTRL and NEG_CTRL from the shift register U3 was using to that task.
They are set or reset in epd_poweron and epd_poweroff. Seems they are here just for nothing.
Maybe there is a reason they provide the signals but didn't route them.
We'd like to hear about this from the designers

lewisxhe, 2021-01-21:
The schematic has been added.

vke, 2021-01-21:
> The schematic has been added.

Please add also PCB  3D model file (STEP?) exported from Altium where PCB was designed

Kabron287, 2021-01-21:
It seems to me that correct naming for this position should be VBAT. Am I right?
[image]

fdufnews, 2021-01-21:
I think your wrong
VBAT is coming throught Q5 and is applied to the "5V" net, when there is no power on USB plug otherwise VBUS is present on "5V' net.
From "5V" net, VBAT (or VBUS) is applied to the BAT net when Q3 is on. And Q3 is on when VBUS is present or PWR_EN is active.

Kabron287, 2021-01-21:
OK, but net BAT on Q3 drain is connected only to R29 and nothing else.
Maybe some nets are not named or some another mistake?

Kabron287, 2021-01-22:
So, the only purpose of the Q3 is to connect battery to ADC divider R29/R32 in order to reduce leak current while device is in deep sleep. But, look at R35. It is permanently connected to battery and consumes ~40uA. It'll be better to change R35 to ADC divider which lower consumption at least twice.

lewisxhe, 2021-01-25:
I have updated the explanation, the voltage divider is not always enabled, only when the operation screen is turned on.

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/b0d619d5fc610bb522ccb7fcf263670c6f8f999a/examples/demo/demo.ino#L113

closed 2022-05-24 by liangyingy