Xinyuan-LilyGO/LilyGo-EPD47 #19: Disable the blue LED

jurecotic, 2021-01-21:
Is there a way to disable the blue led on the back? I would like to rung the board in deep sleep, but the led stays on even in deep sleep.

G6EJD, 2021-01-21:
According to the schematic it is directly across the 3v3 power supply, so no, unsoldering is the only option.

lewisxhe, 2021-01-22:
The light cannot be turned off when the USB is inserted, but the light can be turned off when using a battery.

jurecotic, 2021-01-22:
Ok, how do I turn it off on battery? I'm only using the board on battery and the led is a huge drain. Is there any potential damage by unsoldering it?

G6EJD, 2021-01-23:
I have found no way of switching off the blue LED, deep sleep current consumption is 62mA!
epd_poweroff(); makes no difference to power consumption.

G6EJD, 2021-01-23:
I have  found out why the sleep power may be higher than expected, it is because epd_poweroff(); is not yet implemented.

jurecotic, 2021-01-24:
62mA is useless for anything longterm. Aha, so deep sleep actually has no effect on the display driver board. What should the epd_poweroff() do? Turn off the led and display driver board?

G6EJD, 2021-01-24:
The design is basically good, but after studying the schematic (which is barely legible- needs a better copy) there is no means to turn off the power supply. 
Any power supply control has to Active low to switch it off, because when the ESP32 sleeps it releases all the GPIO pins, the controls would need pull-down resistors for this ideally. GPIO pins go high-z in sleep and effectively get disconnected.

G6EJD, 2021-01-24:
Also there is no means to turn off the Blue LED unless there is some ‘hidden’ power supply off ability, but the schematic has no hint of that, well there is a ver light grey section on the left, again illegible, so can’t say.

I've also determined some of the connectors are incorrectly marked and need correcting.

Display is very fast and durable with the ability to do partial refreshes, so very versatile.

The 16-grey scales works well.

jurecotic, 2021-01-24:
The original EPDiy library also has a method called epd_deinit() which is supposed to be used in conjunction with deep sleep of the esp. 

https://epdiy.readthedocs.io/_/downloads/en/latest/pdf/

lewisxhe, 2021-01-25:
I have updated the instructions for turning off the blue LED, please see
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/b0d619d5fc610bb522ccb7fcf263670c6f8f999a/examples/demo/demo.ino#L114

G6EJD, 2021-01-25:
I’ve looked at the edp_driver.h and these functions are declared / defined:
void epd_poweron();
void epd_poweroff();
void epd_poweroff_all();

But there is then no implementation in edp_driver.c so in effect they doing nothing.

I’ll try the change though in case I’m missing something.

G6EJD, 2021-01-25:
OK, I’ve found the functions now, than you.

G6EJD, 2021-01-25:
epd_poweroff_all(); reduces the supply current down to 275uA, that's a huge difference from the 62mA I measured earlier.

With that result, a 2600mA 18650 battery will last for 394-days if the board did nothing, but more practically, with a 1-hour wake-up and 2-secs of ESP32 activity to update the display, the expected battery life is a respectable 308-days - that's excellent.

This board has a lot of potential.

jurecotic, 2021-01-25:
Works great now. Can we disable the light altogether?

G6EJD, 2021-01-25:
Yes use 'epd_poweroff_all();' and when on battery the blue LED goes off, remains ON whilst USB power is applied, buts that's generally OK.

jurecotic, 2021-01-25:
Yes I know, but it is still on and wasting precious energy while updating the display. Ideally it would remain off permanently...

G6EJD, 2021-01-25:
Get out a soldering iron and remove it, would only take a few seconds. It’s wired to the power rail.

closed 2022-09-02 by liangyingy