Xinyuan-LilyGO/LilyGo-EPD47 #20: Quick start guide not complete

idanov, 2021-01-23:
Hello :wave:  I have followed the quick start guide, but when I try to upload the demo example, I get EPS IDF errors:

```
/home/user/snap/arduino/current/Arduino/libraries/LilyGo-EPD47-master/src/i2s_data_bus.c:5:29: fatal error: esp_idf_version.h: No such file or directory
 #include <esp_idf_version.h>
                             ^
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.
```

I am using Ubuntu 20.04 and installed the Arduino IDE snap. Should I install ESP-IDF too and how to get the demo to work with Arduino IDE?

mol21, 2021-01-23:
Hi, I had the same problem on MacOS 11.1 and platformio with VSCode.
I just deleted or commented out the #include <esp_idf_version.h> lines in the code (it is in multiple files) and it compiles and works.

Nyessss, 2021-01-23:
Hi, I try this too and it worked.
Just delete lines associated with esp_idf_version.h (and if loops) in
/src/i2s_data_bus.c & /src/rmt_pulse.c.

idanov, 2021-01-23:
I found a workaround by installing the dev version of https://github.com/espressif/arduino-esp32. Nevertheless, the step for installing it is missing from the guide and it's probably worth adding it.

lewisxhe, 2021-01-25:
Due to the merged request, using the old version of esp32 core will cause compilation errors

closed 2021-01-25 by lewisxhe