Xinyuan-LilyGO/LilyGo-EPD47 #24: Where can I find the Weather Example?

sthufnagl, 2021-01-27:
Hello,

a lot of screen dumps show a weather example. May I ask "Where can I find this example?"
I know that  epdiy has a weather example, but this dedicated to a different (kindle?) 9.7 " display.
Thanks in advance

BTW, great device with a lot of potential!
:-)

G6EJD, 2021-01-27:
Contact me via g6ejd.dynu.com or dsbird.org.uk and I’ll send you copy, I have just finished a conversion to this display, finished testing yesterday.

sthufnagl, 2021-01-27:
Great work! It works out of the box and look very, very good.
Thx

G6EJD, 2021-01-27:
Good to hear, I think this can be closed now.

sthufnagl, 2021-01-27:
Works...Thx

closed 2021-01-27 by sthufnagl

Jozer99, 2021-01-31:
Any chance I could get a copy?  Are you going to be  releasing a version for this dev board on your git page to go along with the versions for other devices?

G6EJD, 2021-01-31:
Yes, please send me an email via the link above, I’m still improving and testing the sleep function and working on 4-but icons, the version I have now will appear here and my GitHub soon.

mengszym, 2021-02-16:
Hello, after uploading the water program to epd47 after refreshing the screen, the gray background remains.  how to fix it ?  Thank you in advance for your help and best regards