Xinyuan-LilyGO/LilyGo-EPD47 #25: GPIO pin mapping inconsistency between silk screen and picture

markbirss, 2021-02-02:
https://imgaz.staticbg.com/images/oaupload/banggood/images/7C/F3/09005c65-7fbf-423d-bb64-2eca9abe5f16.jpg.webp

The picture shows

V 21 22 G 
for the last port

silk screen shows
VCC 14 15 G

Cant get either of these too work ?

lewisxhe, 2021-02-04:
Ok i will give them feedback, thank you

closed 2021-02-04 by lewisxhe

G6EJD, 2021-02-04:
Today I’m going to try out the pins to see if they can be controlled, I don’t envisage any issues but it won’t take long to find the GPIO ports in use. I note that the silk screen says 12.13 and 14,15 with 21,22 being used for the touch interface.

markbirss, 2021-02-04:
@G6EJD 

Can you confirm the layout of the GPIO pins for the connectors and also the direction for clarity ?

G6EJD, 2021-02-06:
I used the bme280 sensor, it needs address 0x77, Wire,begin(14,15);
Plugged it into the port marked 14,15 works OK

I used the bme280 sensor, it needs address 0x77, Wire,begin(12,13);
Plugged it into the port marked 12,13  IT DOES NOT WORK

The problem is associated with Pin-12, which if it has an external pull-up of any time (the sensor) prevents the ESP32 from booting, and this is what I see happening. Pin-12 was a bad choice and should be changed in later releases of the board.

markbirss, 2021-02-06:
@G6EJD 

Ok, i will have a look again since i want to enable Flow Control over serial and require 2 more GPIO

G6EJD, 2021-02-06:
If you make GPIO-12 an output for the handshake line and that line is pulled low during boot by the external circuit then you'll be OK. A measurement of your disconnected external circuit on the RTS/CTS lines will confirm if you will have an issue.

GPIO-13 is free for use either mode of I/O.

markbirss, 2021-02-06:
@G6EJD 

Thank you for your guidance,

I will see how that works out, I will use boot_delay in Raspberry to delay the boot up to give the ESP a change (if i use GPIO 12 and have a issue pulling it high)

G6EJD, 2021-02-06:
Yes if the RPi leaves GPIO pins high-Z until allocated that should work. If you have a DVM simply measure the voltage on your chosen pin for RTS/CTS then you'll know, any high >2v will be an issue for GPIO-12

We now know the silk-screen pin labelling is correct.

markbirss, 2021-02-06:
@G6EJD 

Thank you

markbirss, 2021-02-06:
Ended up using 12 - RxD ,13 - TxD, 14 - CTS and 15 - RTS and a boot_delay of 2 seconds for now

I dont need to connect anything else to the ESP32 as my USB keyboard works from the Raspberry Pi

[image]

G6EJD, 2021-02-06:
Good to see it working.

Hendrikxs, 2021-09-15:
New user here,
Can you use the touch panel (GPIO 13, 14 and 15) at the same time?
Regards,