Xinyuan-LilyGO/LilyGo-EPD47 #26: Not working with connected battery

BatsIhor, 2021-02-06:
Hi there, 

I use my for a few weeks and finally decided to switch from USB to battery and it's not working. 
I checked voltage on battery and it's all good, 3.7v but once connected board doesn't show any signs of life, no lights, nothing. 
Once connected a USB cable it starts to work, blue LED is ON, green is not lighting up. 
Once battery is disconnected blue and green LED is ON

How can I troubleshoot this issue?

Thank you
Ihor

G6EJD, 2021-02-06:
What type of battery connector do you have if 18650 have you noted that the battery cannot be the flat protected type, although I use that type as there is no connection, so put the battery in and push the connectors onto the cells to see if that works, if so remove the battery and gently bend in the connector ends.or use a button type 18650.
If using the SH connector then what polarity are you using as there is no standard. Looking at the rear of the board with the 18650 battery position at the bottom the + of the SH connector is on the Left

BatsIhor, 2021-02-06:
I use SH and you are right polarity was not correct, I changed it and all works fine. 
Thank you very much, was not aware that polarity might be different.

closed 2021-02-06 by BatsIhor

G6EJD, 2021-02-06:
Good to hear it's working now.
+ and - should have been on the silk-screen.

giladaya, 2021-12-03:
@G6EJD Thanks for that tip!
You helped me solve a problem with a new battery.

G6EJD, 2021-12-03:
If only they had checked other boards before designing the pcb!