Xinyuan-LilyGO/LilyGo-EPD47 #27: LilyGo-EPD47/examples/drawImages/ - what is used to create the .h files

markbirss, 2021-02-11:
For the examples what was used to create the .h files
Howto convert jpg to .h ?

G6EJD, 2021-02-11:
There is a script have you checked out the image converter variant?

markbirss, 2021-02-11:
@yes i have this one for example
https://javl.github.io/image2cpp/

but i get the same picture display smaller 4 times next each other?

G6EJD, 2021-02-11:
No need copy the file you want .jpg into a folder and copy imgconvert then run it using g the example provided imgconvert source.ext destination.ext then copy the resultant .h file to your sketch

G6EJD, 2021-02-11:
1.  On Windows PC install Python 3.91 or later, find it here: https://www.python.org/downloads/

2.  Install Python

5.  Create an accessible folder e.g. EPD47 e.g. C:\EPD47   ***Note this folder must be accessible to the cmd pompt
    A better location might be in your Documents folder usually located at C\Your-Username\Documents, example:
    Directory of C:\Users\david\EPD47

6.  Copy your source .JPG images to the folder

7.  Transfer the following files to your EPD47 folder - imgconvert.py and fontconvert.py they are located in the EDP47 Library script folder

8.  Install Python 3.91, start a Windows cmd prompt, and chage directory to the folder EPD47:
    cd C:\Users\david\EPD47

9.  To create an image for display, enter this at the command prompt:
    imgconvert.py -i sourceimagename.jpg - n requiredname -o outputfile.h

10. Example convert sunny.jpg to sunny.h
    imgconvert.py -n sunny -i sunny.jpg -o sunny.h

11. The image file is now created, include the file in your code like this #include "sunny.h"

markbirss, 2021-02-11:
I dont use Windows

but i see it on this page thanks

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/scripts/README.MD

closed 2021-02-11 by markbirss