Xinyuan-LilyGO/LilyGo-EPD47 #28: epd_draw_pixel not working

nellydocs, 2021-02-15:
i tried using 

```
void epd_draw_pixel(int x, int y, uint8_t color,
                    uint8_t *framebuffer);

```
from epd_driver.h

i tried it using
`
epd_draw_pixel(500,200,0,framebuffer);
`
and 
`epd_draw_pixel(500,200,255,framebuffer);`

both lines didnt work unf.

i am using the touchtest example code where i removed the center text.
a snippet where i added the line
`  epd_poweron();
    epd_clear();
 
    //Draw Box
    epd_draw_rect(600, 450, 120, 60, 0, framebuffer);
    cursor_x = 615;
    cursor_y = 490;
    writeln((GFXfont *)&FiraSans, "Prev", &cursor_x, &cursor_y, framebuffer);

    epd_draw_rect(740, 450, 120, 60, 0, framebuffer);
    cursor_x = 755;
    cursor_y = 490;
    writeln((GFXfont *)&FiraSans, "Next", &cursor_x, &cursor_y, framebuffer);
    
    epd_draw_pixel(500,200,0,framebuffer);
    
    Rect_t area = {
        .x = 160,
        .y = 420,
        .width = lilygo_width,
        .height =  lilygo_height
    };
    epd_copy_to_framebuffer(area, (uint8_t *) lilygo_data, framebuffer);

    epd_draw_rect(10, 20, EPD_WIDTH - 20, EPD_HEIGHT / 2 + 80, 0, framebuffer);

    epd_draw_grayscale_image(epd_full_screen(), framebuffer);
    epd_poweroff();` 

what am i doing something wrong?
could someone correct me please or prive a small snipped?

thanks

martinberlin, 2021-03-01:
Looks ok are you sure is not drawing the pixel over another black element and you don’t see it?
Try to draw 3 or 4 and do it on gray (255/2 or alike)
I would also start with an empty screen to draw a pixel like this.

martinberlin, 2021-04-24:
@nellydocs unless Lilygo has the will of updating this one day ( Check #33 )

I highly recommend to use this repository:
https://github.com/vroland/epdiy

That has an upgraded version of the same base library used here. This also supports software rotation that is here not implemented and comes with examples. Works for ESP-IDF Framework and also Arduino-esp32.

Alternatively if you want to use Espressif IDF and C++ I have a [component called Cale-idf](https://github.com/martinberlin/cale-idf) that fully supports this epaper and it's touch component.

closed 2024-12-28 by github-actions[bot]