Xinyuan-LilyGO/LilyGo-EPD47 #30: Cannot get the Touch example to work right

martinberlin, 2021-02-28:
Hi there @lewisxhe 
I'm trying to implement the [touch for this epaper in my IDF component](https://github.com/martinberlin/cale-idf/issues/34) (For parallel is also based on EPDiy)

I started simply adding this repository as a library and compiled first the grayscale to confirm it works good. Connected the touch like described from Lilygo in a video. And then I copied the touchtest example and compiled it. It renders the LilyGo logo with the text and the Prev | Next buttons but touch seems way off. 

For example if I put a Serial output in the loop:

```c++
void loop()
{
    uint16_t  x, y;
    if (digitalRead(TOUCH_INT)) {
        if (touch.scanPoint()) {
            touch.getPoint(x, y, 0);
            y = EPD_HEIGHT - y;

            Serial.printf("X:%d Y:%d\n");
         // do the rest
    }
}
```

**All what I get in Serial is this:**
Started Touchscreen poll... -> I2C seems to be working correctly and touch FPC also alright!
X:1073422222 Y:463
X:1073422222 Y:470
X:1073422222 Y:473
X:1073422222 Y:563
X:1073422222 Y:633

As you can see the Y seems to retrieve the right coordinate since I touch from the middle to the right till the area where the Prev | Next buttons are. But the X is giving a super high number that does not change at all when moving up. Can someone shed some light on this?

As I side topic I do not understand why Lilygo purchases some touch controller from a provider, but is **not able to supply either the Vendor specifications or a proper Technical specs PDF** where we developers can properly get informed about the touch events and more. Touch is not only about Tapping and getting X & Y Coordinates, some touch interfaces also have gestures and return different events like Tap, Drag, Press, Release etc. 

Second thing I don't understand correctly is why this is not like a standard I2C implementation as we can see for example in [this Goodisplay FocalTech component initialization](https://github.com/martinberlin/cale-idf/blob/master/components/FT6X36-IDF/FT6X36.cpp#L36).
Usually when starting communication with I2C, both temperature sensor and other touch components first thing they do is to check that the Vendor ID and Chip ID match. Otherwise might be hitting the wrong component. And that's what I see on most I2C, but here is simply assuming is like this, can work is no problem but it shows a bit that you are implementing something that you don't understand completely. This lack of information summed to the fact that I cannot even get the Lilygo touch test demo working correctly makes the touch unusable for my intents right now.

If someone can help I will be really happy to get this moving forward and implement this differently using proper Rising edge interrupts instead of looping all the time over, that is first not the right way to do it, and secondly higher CPU intensive if you application does other things other than touch. Touch should be something that runs smooth on the background and is just triggered when TOUCH_INT pin goes HI  (This is also uncommon, on other touch interfaces, same as Chip select INT is low when receiving an event) But is ok I'm just interested in getting this to work and getting the full technical information about the controller.

lewisxhe, 2021-03-01:
1. I can't add more functions to the touch, because the supplier only provides me with a simple sample program, and did not give me a data manual. I asked them for it, but they did not provide it.
2. I have tested it and it will not fail to read
3. I can provide you with the sample program that the supplier gave me to facilitate your research, but it is not helpful to me. 

[5_5_Touch.zip](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/6059097/5_5_Touch.zip)
[L58数据手册V1.0.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/files/6059098/L58.V1.0.pdf)

martinberlin, 2021-03-01:
I understand. But to implement touch you need at least press down and lift up events.
Without that is completely useless. Because otherwise there is no real press of a button. 
you press and hold, then you repeat same button press multiple times.
I’m sorry but if this L58 has not that then it was simply a poor choice for a touch controller. I will try to translate the PDF and sniff all that comes per I2C but as a consumer I shouldn’t need to reverse engineer your products to get them running.
In the code I see just X and Y returning via I2C. If this is really the touch controller for a mobile screen it should have events, they need to be somewhere . Thanks for your time

lewisxhe, 2021-03-01:
I feel helpless ! This is not my choice.

martinberlin, 2021-03-01:
It's no problem, I just want to help, so maybe we can get the events somehow. But if we can not then at least this should be in the product description explained. Otherwise it can make other developers helpless too ;(

martinberlin, 2021-03-01:
Updated my touch examples and the wiki. Added “tap simulation” with a timeout. Still does not work 100% correctly like having press and lift up evens but is more usable:
https://github.com/martinberlin/cale-idf/wiki/Model-parallel-ED047TC1.h#touch-research

@lewisxhe I will sniff the I2C and try to discover if has events. But next time if the provider is not able to deliver proper documentation/examples then its simply not a good choice even if the price quality is great (looking at least)

lewisxhe, 2021-03-02:
I have explained the problem to LilyGo, and the touch panel should be replaced later. Thank you for your contribution.

martinberlin, 2021-03-06:
Waiting for another panel since mine is broken. Lilygo sent me a proposed solution so when it arrives I will test it and document it here. It seems that comes at least a press down /lift up event.
Added it here in my component but I need to wait to test it:
https://github.com/martinberlin/cale-idf/commit/fbf637181b3afd31e7f1d04637cd730ffb610d13

fonix232, 2021-03-24:
@martinberlin unfortunately this sort of poor component choosing is not unique to LilyGO. I see this in almost every custom design developer kit coming from China. Well tested components and systems are often canned for undocumented, "cheaper" counterparts, most of the time for no good reason.

For example, the M5Paper (M5Stack's version of the EPD47) has a bunch of questionable choices. The EPD is driven by an IT8951, an SPI-connected microcontroller (with lots of great capabilities, by the way!), which ends up saving a few pins only to cause a permanent 100-130mA drain when on. The AXP192 PMU is replaced with a lesser controller that has much less documentation, and can't even report events like charger connected, or charging complete. And one could go on.

To me, both of these boards seem like they were put together by a handful of teams with little to no communication with each other, often the solution just getting "slapped on", with little regards to compatibility with the rest of the system. 

And then we end up with devices that look good on paper, but have fatal design flaws, issues that cannot be fixed in software. I for one would rather have a good hardware design that is a bit more expensive and has little software support initially, than to have a device that cannot be fixed due to faulty hardware.

martinberlin, 2021-03-24:
I agree on your points. But I must say that this epaper proves to be quite nice, both in resolution, price and quality. I like it so far.
I'm still waiting for my new touch. 
One point I dislike: This touch FPC connectors on the PCB are not the best quality. I understand you are not suppose to plug in and out the touch many times, but mine broke after just a few times. And when this tiny things brake it's impossible to repair.

The issue is that once you stick the touch, if it brakes the connector, you have a half usable epaper. 
I will update this when the touch arrives and I still have to confirm if it delivers press/release events that are necessary to detect the "TAP" event. Without that is very hard to make an usable button ;)

martinberlin, 2021-04-24:
@lewisxhe touch is now updated and supporting events in [Cale-idf epaper component](https://github.com/martinberlin/cale-idf/releases/tag/v1.0.5) and my ESP-IDF component also:
https://github.com/martinberlin/FT6X36-IDF Supports L58 touch. 

Proof-of-concept video:  https://twitter.com/martinfasani/status/1384244324922105864

closed 2021-04-24 by martinberlin