Xinyuan-LilyGO/LilyGo-EPD47 #32: Waveform File

vroland, 2021-03-24:
Hi, 
Recently, I added support for vendor wavforms to epdiy. These are special files that describe how the display must be driven for accurate grayscale.
This allows to make arbitrary grayscale-to-grayscale transitions, which is not possible with the version of epdiy this library is based on at the moment. There are some other improments as well, especially to the user-facing API. So incorporating them or finding a way to keep this fork more up to date should be discussed in a separate issue.
You have sent me the fitting waveform file via e-Mail already, but it would be nice to have it publicly available for anyone who needs it. That would allow me to add the appropriate header to epdiy directly, which would make it way easier for people to get use out of this file.
So: Could you add the V220_C244_BA5701_ED047TC2U1_TC.wbf waveform file to the repository?

Regards,
Valentin

markbirss, 2021-04-04:
Please upload the **V220_C244_BA5701_ED047TC2U1_TC.wbf** waveform file for all of us

LilyGO, 2021-05-06:
@vroland Ok.
Sorry, I haven’t logged into GitHub recently
Thank you for your efforts, I should reply to you as soon as possible

LilyGO, 2021-05-06:
@vroland Have you submitted your pull request? So that I can approve your merge request.
Contributors will also show your name

vroland, 2021-05-06:
Making a pull request for the waveform implementation only is unfortunately not trivial, because a lot of refactoring happened in epdiy since then. Maybe we should discuss this in https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/33 instead?
Also, publishing the waveform file would be a prerequisite for generating a suitable waveform header.

hacksics, 2021-05-31:
I've started using epdiy instead of this fork and it makes sense to use the base repository as it has lot of new features and updated frequently. 

But what I have noticed was the image colors are not accurate as EPD47 when using native epdiy. Is this due to using the EPD_BUILTIN_WAVEFORM file instead of vendor waveform? 

@vroland as suggested by @LilyGO  can you create a pull request by just adding the raw waveform file so we can use it to generate header file using waveform_hdrgen.py ?

vroland, 2021-05-31:
@hacksics Because I'm unsure of the copyright situation, I would prefer that @LilyGO publish the file, since I got it from them.

hacksics, 2021-06-01:
Ok makes sense. @LilyGO @lewisxhe  can you please chime in..

Jason2866, 2021-10-03:
+1 @LilyGO for uploading **V220_C244_BA5701_ED047TC2U1_TC.wbf**

hacksics, 2021-10-21:
@vroland can you please integrate this in to epdiy repo ?

entozoon, 2021-11-30:
@hacksics [PR created](https://github.com/vroland/epdiy/pull/129)

closed 2024-12-27 by github-actions[bot]