Xinyuan-LilyGO/LilyGo-EPD47 #33: New Epdiy Features

vroland, 2021-03-24:
Hi, 
as mentioned in a previous issue, epdiy, the driver this library is based on, has made some progress since this repository was created. Notable changes are:
* A high-level API that makes allows to simply write to a framebuffer and let the driver update the display, no need to know about different update modes / calculating partial updates.
* Removed namespace clashes with Adafruit GFX
* Support for vendor waveforms, allowing arbitrary grayscale-to-grayscale transitions
* Support the the arduino IDE. Currently, this requires to install a board definition file because of limitations of arduino-esp32, but since this library only supports one board this process could be simplified.
* The driver can be configured to use significantly less memory at the expense of performance, which is necessary when using wifi and bluetooth together.
* The documentation has been updated to make things like image and font generation clearer.

Thus, I thought it would make sense to discuss how these changes could be best integrated here, to take advantage of the development effort happening in both places. Maybe this library could wrap epdiy and add device-specific examples and extensions?

martinberlin, 2021-03-28:
Thanks for the great update @vroland 
As I commented in #9 I would like to add software rotation here too but I'm not going to do it with the existing old copy of EPDiy.
I would like to do it correctly and link EPDiy as a base library in the platformio.ini or at least if that is not the preferred way by Lilygo to have an epd_driver copy of the latest version.
But I would favour to use EPDiy as an arduino-esp32 dependency and to set the Epaper / Board defines directly using build_flags in platformio.ini (Or any other constant that needs to be defined in order for this to work correctly)

Lorysmus, 2021-05-13:
Hi there I play with the Lilygo EPD47 with the Epdiy library because of the rotation feature which works very well now! Great work Martin... Unfortunately I did not find out the way how to reduce the power consumption to minimum in deep sleep mode with this library - Do you have any experience? I tried to use epd_deinit() but without any effect.

martinberlin, 2021-05-13:
Thanks @Lorysmus much appreciated. I actually didn’t check consumption with fine detail yet.

@vroland do the parallel epapers also have a deepsleep mode?
So far for many demos like weather all the ESP32 is going to sleep till next refresh but I do not have much experience with parallel yet, just understand parts of the code since I added the rotation.
By the way do you have measurements in consumption, running same program, with and without epaper connected? So far I didn’t try that.

vroland, 2021-05-14:
With the v5 board, I managed to get 13uA deep sleep current. But I did not test with the LilyGo board, yet. I guess a dedicated deepsleep demo would be useful, but I did not get arround to that, yet.

DavidM42, 2021-05-14:
The example I added should demo low power usage of the lilygo epd 4.7 inch fairly well.
The code I use which formed the base for my example is sitting in deepsleep for days or weeks on battery now.
I use both `epd.poweroff()` and then the typical esp deepsleep start

AndreKR, 2021-11-17:
Just for reference because I didn't immediately understand it.

Both the LilyGo and the Epdiy board use a shift register to make various hardware settings. The library function `epd_poweroff_all()` just sets all outputs to LOW to disable the power supply for minimum power consumption:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/37e869141f7469b3438a3434e9d5ff60cc3caeb8/src/ed097oc4.c#L147-L151
From the schematic I think the crucial bit is PWR_EN, which in the code is still (taken from the epdiy source) called `ep_scan_direction`:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/37e869141f7469b3438a3434e9d5ff60cc3caeb8/src/ed097oc4.c#L53-L68

This is now also taken into account by the original Epdiy project:
https://github.com/vroland/epdiy/blob/84898485658af384a9b71148c9294daf56c96abd/src/epd_driver/config_reg_v2.h#L69-L84

> ```c++
> static void cfg_poweroff(epd_config_register_t *cfg) {
> #if defined(CONFIG_EPD_BOARD_REVISION_LILYGO_T5_47)
>   // This was re-purposed as power enable.
>   cfg->ep_scan_direction = false;
> #endif
>   // POWEROFF
>   cfg->pos_power_enable = false;
>   push_cfg(cfg);
>   busy_delay(10 * 240);
>   cfg->neg_power_enable = false;
>   push_cfg(cfg);
>   busy_delay(100 * 240);
>   cfg->power_disable = true;
>   push_cfg(cfg);
>   // END POWEROFF
> }
> ```

closed 2024-12-26 by github-actions[bot]