Xinyuan-LilyGO/LilyGo-EPD47 #34: MicroPython?

astranome, 2021-04-06:
Hi, thanks for such a good project!
Just out of curiosity - how difficult (or feasible) is to port your driver under Micropython (esp32 in particular)?

martinberlin, 2021-04-24:
I made a search and seems not to be any Micropython project so far to support parallel epapers on the ESP32. 
Would be great to have something but the task seems complicated. If you are interested on how to do it, I highly recommend you to study [EPDiy](https://github.com/vroland/epdiy/) that is the C base library to support this and many other parallel epaper displays.

ondiiik, 2021-05-20:
You can try [this one](https://github.com/ondiiik/micropython-twatch-2020). It has been originaly created as port for [TWatch-2020](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library), but after some time I also added support for some other boards. Board name do build correct variant for this EPD is **LILYGO_T5_EINK_47**. There is module `epd` with some basic support (can be used with `framebuffer` module in mode `GS4_HMSB`).

Just note that there is some issue related to EPD and WiFi. I don't know the real reason, but WiFi can not be used together with EPD. I resolve it by 2 staged boot. When board is woken up from deepsleep, thewn WiFi is connected and all needed stuff is loaded into flash (using littlefs to prevent from erly flash demage caused by often write attempt). When all jobs requiring WiFi are done, then board is rebooted and after reboot it print out all needed on display and go to deepsleep.

gmorell, 2022-01-12:
@ondiiik Do you have a workaround for the wifi/epd issue or an example of the two stage boot in micropython? I'm curious to see how that works.

ondiiik, 2022-01-12:
> @ondiiik Do you have a workaround for the wifi/epd issue or an example of the two stage boot in micropython? I'm curious to see how that works.

Unfortunately only workarround I have is to boot twice. First boot activates WiFi and proceed all required communication. Board is then rebooted and results displayed on EPD. After that board goes to deep sleep. To distinguish if you shall activate WiFi or draw on screen you can use `machine.reset_cause()`. When you get `machine.DEEPSLEEP`, this means that last boot was about drawing display.

You can have a look [here on project building meteostation](https://github.com/ondiiik/meteoink/blob/lilygo_t5_eink47/qt/app.py#L102) over this board.

LilyGO, 2022-04-07:
Micropython is currently supported, please visit [lilygo-micropython](https://github.com/Xinyuan-LilyGO/lilygo-micropython.git).

This issus will be closed, all about micropython, please discuss at [lilygo-micropython](https://github.com/Xinyuan-LilyGO/lilygo-micropython.git).

closed 2022-04-07 by LilyGO