Xinyuan-LilyGO/LilyGo-EPD47 #35: error compiling arduino

rolohaun, 2021-04-07:
I get this error when compiling in Arduino

sketch\demo.ino.cpp.o:(.literal._Z5setupv+0x6c): undefined reference to `writeln'
sketch\demo.ino.cpp.o: In function `setup()':
C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\demo/demo.ino:72: undefined reference to `writeln'
C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\demo/demo.ino:86: undefined reference to `writeln'
C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\demo/demo.ino:89: undefined reference to `writeln'
C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\demo/demo.ino:93: undefined reference to `writeln'
sketch\demo.ino.cpp.o:C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\demo/demo.ino:95: more undefined references to `writeln' follow
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
 Used: C:\Users\Ron\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

G6EJD, 2021-04-07:
Do you the screen driver installed?

rolohaun, 2021-04-07:
ok I added the epdiy library to the demo sketch and that solved the error however now it's asking for a display type

In file included from C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver/include/epd_driver.h:11:0,
                 from C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver.h:4,
                 from C:\Users\Ron\AppData\Local\Temp\arduino_modified_sketch_548536\demo.ino:1:
C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver/include/epd_internals.h:55:2: error: #error "no display type defined!"
 #error "no display type defined!"
  ^
Multiple libraries were found for "epd_driver.h"
 Used: C:\Users\Ron\Documents\Arduino\libraries\epdiy-master
 Not used: C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master
Multiple libraries were found for "SD.h"
 Used: C:\Users\Ron\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.

G6EJD, 2021-04-07:
Try one of the other examples eg drawimages

rolohaun, 2021-04-07:
same error

In file included from C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver/include/epd_driver.h:11:0,
                 from C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver.h:4,
                 from C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master\examples\drawImages\drawImages.ino:9:
C:\Users\Ron\Documents\Arduino\libraries\epdiy-master\src/epd_driver/include/epd_internals.h:55:2: error: #error "no display type defined!"
 #error "no display type defined!"
  ^
Multiple libraries were found for "epd_driver.h"
 Used: C:\Users\Ron\Documents\Arduino\libraries\epdiy-master
 Not used: C:\Users\Ron\Documents\Arduino\libraries\LilyGo-EPD47-master
exit status 1
Error compiling for board ESP32 Dev Module.

rolohaun, 2021-04-07:
I was using the epdiy library direct from the github and not from this git. I cleaned everything out and now it seems to work

G6EJD, 2021-04-07:
Close the issue then?

closed 2021-04-07 by rolohaun

skychecker, 2022-12-07:
> I was using the epdiy library direct from the epdiy/git and not from this git. I cleaned everything out and now it seems to work

Could you please explain more details about how had you fixed the issue ? I have the absolute same problem. Thank you.