Xinyuan-LilyGO/LilyGo-EPD47 #39: Weather Application Missing

johnwargo, 2021-04-22:
Is there a reason the weather application isn't included in this repo?  Can you please add it?

martinberlin, 2021-04-24:
I highly recommend to use this repository:
https://github.com/vroland/epdiy

That has an upgraded version of the same base library used here. There is also a full weather example that can be compiled using ESP-IDF framework:
https://github.com/vroland/epdiy/tree/master/examples

[image]

But the repository also can be used with Arduino-esp32, you can take a look at this proposed example: 
https://github.com/vroland/epdiy/issues/22

guyou, 2021-04-25:
> Is there a reason the weather application isn't included in this repo? Can you please add it?

IMHO it is a copyright/licence issue. See [discussion on G6EJD's repo](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/135)

@LilyGO I think you have to revise the PDF tutorial.

johnwargo, 2021-04-25:
Created #42 to add a readme to this project referring to the weather app so everyone can easily find it directly from this repo. 

@LilyGO if you send me the source file for the PDF, I can clean up the file and produce a new version with the link pointing to the repo as well (and move the PDF file to the new Weather folder I just added).