Xinyuan-LilyGO/LilyGo-EPD47 #46: Load Image from UEL

E-TV, 2021-05-23:
Hi!
I have a project in mind with the Spotify API. I want to make some kind of music-control. Since I have this E-Paper-Display I thought it would be interesting to display the picture of the song. With the Spotify-API I can already get the URL of the Song-Picture (e.g. https://i.scdn.co/image/ab67616d00001e02255e131abc1410833be95673). But of course I can't download the Song-Pictures of millions of songs to the EPD47. So I have the question: Is there a way to show pictures from an URL?

Thanks in advance

otosan-maker, 2021-05-31:
> Hi!
> I have a project in mind with the Spotify API. I want to make some kind of music-control. Since I have this E-Paper-Display I thought it would be interesting to display the picture of the song. With the Spotify-API I can already get the URL of the Song-Picture (e.g. https://i.scdn.co/image/ab67616d00001e02255e131abc1410833be95673). But of course I can't download the Song-Pictures of millions of songs to the EPD47. So I have the question: Is there a way to show pictures from an URL?
> 
> Thanks in advance
> Bastian
> 
> BTW: Do we already have a LILYGO-Discord? I think that would be quite helpful

I have adopted a function I have found around to be used to display a BMP image. I suggest you create a WebService then download the image, transform it to BMP and download it to the e-paper...
It has to be easy ... maybe four hours :-D

martinberlin, 2021-10-03:
Please give a try to my JPG example in the component EPDiy 
https://github.com/vroland/epdiy
That is the father's repository of this project.

liangyingy, 2022-09-02:
You can try [wifi_sync](../tree/master/examples/wifi_sync) example.

closed 2022-09-02 by liangyingy