Xinyuan-LilyGO/LilyGo-EPD47 #47: Screen goes blank when battery is connected

hacksics, 2021-05-27:
I have a strange issue. When I connect the 18650 battery, screen goes blank however blue LED is on. I've tested if ESP32 getting 3.3v and confirmed that its getting 3.3v. 

What could be the issue? 

Also can I solder wires to PH connector and use instead? Does anyone know they use different circuitry ?

benkokes, 2021-12-10:
I also have this problem. The unit works when plugged into USB, but if you install a 18650 battery, the screen slowly goes blank. There are a few missing passive components next to the + terminal, could that be the issue?  I have not removed any components from this board.

Schnabulation, 2022-01-31:
I have the same problem. I can confirm the battery is of type 18650 and fully charged. How can I fix that?

G6EJD, 2022-01-31:
Are you sure the battery is making contact with the battery terminals you need to bend them in quite some way to ensure contact, especially if you have the flat top 18650 type of battery.

Schnabulation, 2022-01-31:
Thanks for the tip. Yes, I have bend the terminals pretty far so that the battery makes contact. I can also verify that all ESP32 functions (such as GPIO ports, WiFi, etc.) work flawlessly - it's only the display that does not work.

daernsinstantfortress, 2022-12-20:
@Schnabulation Did this issue ever return for you? I have two of these boards, and both are now displaying the same symptoms - blank display when battery inserted (with or without external power), but work fine when just powered from USB. I've played with the battery terminals, but to no avail.

Schnabulation, 2022-12-21:
Actually this was fixed by bending the terminals and never returned.

daernsinstantfortress, 2022-12-21:
> Actually this was fixed by bending the terminals and never returned.

@Schnabulation Thanks (and I've just realised that you also responded earlier on the other thread too!)

Today, I soldered on an 18650 holder to the battery terminals and, lo and behold, it all works fine now:

[image]

I can't work out whether the battery clips are just not very good conductors, or the tension of the battery is upsetting something else on the board, but desoldering them and replacing with this is an easy, quick fix for me and if it turns out reliable (as it seems to be now), then it'll do for me!

mgaman, 2023-04-25:
I also have the same problem. When plugged into a computer via USB all works fine, but if USB plugged into a power bank the screen doesn't work.
After much investigation I found that the problem is Serial I/O. If the code is devoid of any Serial code, even just Serial.begin() then everything works, even from a power bank.
I didn't investigate running from a battery as I don't have one.
Hope this adds some insight.

closed 2024-12-23 by github-actions[bot]