Xinyuan-LilyGO/LilyGo-EPD47 #52: Ghost image when redrawing screen

nicolacimmino, 2021-06-06:
Hi,

I am having an issue that when I display some text on the screen the previously displayed one is visible in negative (whiter than the background).

any hints on what could be wrong?

I always write to a fully cleared frame buffer first:

````
writeln(font, buffer, &x0, &y0, Peripherals::framebuffer);
````
and then draw the full buffer:

````
  epd_poweron();
    epd_clear();
    epd_draw_grayscale_image(epd_full_screen(), Peripherals::framebuffer);
    epd_poweroff();
    memset(Peripherals::framebuffer, 0xFF, EPD_WIDTH * EPD_HEIGHT / 2);
````

See picture below for the issue:

[image]

martinberlin, 2021-10-03:
This a normal effect on this epapers. That's why kindle writes kind of a reverse image when updating. But also in other epapers like my Hisense color phone where o write this lines have the ghost effect. Only using full clear mode that is super slow solves it.

AndreKR, 2021-11-13:
So, what do we need to do fix it, like the Kindle does?

I'm running into the same issue. Here you can clearly see the text "Voltage: 4.94V" because I was running the "demo" example before and you can even see the eyes of the manga girl because I once ran the "drawImages" example on the display:

[image]

I think the reason why you can only see the "Voltage" line and not the other lines from the "demo" example is that they are somehow drawn differently because of partial refresh. Here is a picture with the "demo" example running and you can see how the partially refreshed area has much higher contrast:

[image]

giladaya, 2021-12-20:
I struggled a lot with this issue.

The initial approach I tried was to draw the negative of the previous image before clearing the screen and painting the new image.
This can be done using the `WHITE_ON_WHITE` display mode like this:
`epd_draw_image(epd_full_screen(), frameBuffer, WHITE_ON_WHITE);`

This works, but the downside is that you need to have a copy of the previous image, which is not always possible.

Eventually I came up with this sequence for drawing the new image, which seems to solve the ghosting and give good contrast:
```c++
void drawFrame(uint8_t *framebuffer)
  {
    epd_clear_area_cycles(epd_full_screen(), 2, 80);
    delay(100);
    
    epd_draw_image(epd_full_screen(), framebuffer, WHITE_ON_WHITE);
    epd_draw_image(epd_full_screen(), framebuffer, WHITE_ON_WHITE);
    epd_draw_image(epd_full_screen(), framebuffer, WHITE_ON_BLACK);
    
    epd_draw_image(epd_full_screen(), framebuffer, BLACK_ON_WHITE);
  }
```

Notes:
- It's a bit slow but you can play with the commands to find a balance between refresh time and quality.
- Note that I'm mostly using the display to show pictures (not text), so YMMV.

geiseri, 2022-03-08:
I found the more times I update the screen with the same image the stronger the ghosting is and the more times I need to run the code above.  @LilyGO is there a better waveform that can be used to mitigate this?

closed 2025-02-04 by github-actions[bot]