Xinyuan-LilyGO/LilyGo-EPD47 #56: ESPNow

fugazigames, 2021-08-09:
Using ESPNow in slave mode works perfectly, except when I try to use the writeln function.  It prints one time, then hangs.  If I remove the writeln function it continues to print to the console as expected.  Any help would be appreciated.

void OnDataRecv(const uint8_t *mac_addr, const uint8_t *data, int data_len) {
  char macStr[18];
  snprintf(macStr, sizeof(macStr), "%02x:%02x:%02x:%02x:%02x:%02x",
           mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
  Serial.print("Last Packet Recv from: "); Serial.println(macStr);
  Serial.print("Last Packet Recv Data: "); Serial.println(*data);
  Serial.println("");
  writeln((GFXfont *)&OpenSans24B, data, &temp1_cursor_x, &temp1_cursor_y, NULL);
}

parahooners, 2021-08-16:
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {
  memcpy(&myData, incomingData, sizeof(myData));
 Serial.print("Bytes received: ");
  Serial.println(len); 

  pulses = myData.a; // pulses incoming

}

It's difficult to see with out the whole code espnow is tricky....

I always store the incoming data before presenting it...

liangyingy, 2022-05-24:
Issues not related to this repo

closed 2022-05-24 by liangyingy