Xinyuan-LilyGO/LilyGo-EPD47 #57: Faded display

stefan-sherwood, 2021-08-22:
I am using the LilyGo 4.7" ePaper display (https://www.aliexpress.com/item/1005002617525312.html)

When I got the unit, I plugged it in, and it showed the LilyGo demo app. Everything looked great, contrast, and display.

I followed the instructions [here](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) to install the demo app.

The app compiled and uploaded successfully. It works well, except that the display is quite faded with poor contrast. I tried the other examples and have the same problem. I don't think the unit is defective since it worked fine before I uploaded the demo.

My setup:
**OS:** Windows 10
**IDE:** Arduino IDE v1.8.15 (current as/of 8/22/2021)
**Board:** ESP32 Dev Module
**PSRAM:** Enabled
**Partition scheme:** 16M Flash (2MB APP/12.5 MB FAT)
_[I have tried all of the partition schemes, all with the same result)_

I have also tried with Arduino IDE 2.0.0-beta.10 and get the same result

[image]

stefan-sherwood, 2021-08-22:
Here is a picture of the _grayscale_test_ example. Note that these pictures were taken with a bright light. In normal light, the display is barely visible.

[image]

G6EJD, 2021-08-22:
Has the display been exposed to high levels of UV / sunlight? 
What is/was the ambient temp during testing?
It can help to fill the screen with all black, wait for 180-secs, clear the screen, then all white to depolarise the panel.

stefan-sherwood, 2021-08-22:
Definitely not since I got it. I took it out of the package, plugged it in and it worked perfectly. I plugged and unplugged it a few times and the contrast was perfect every time. Within an hour I uploaded the sketch and it immediately was dim.

I will try what you suggested.

stefan-sherwood, 2021-08-23:
Alas, the depolarization technique did not make any difference. It _looks_ to me like the display may be expecting more bits for grayscale than the driver is providing since everything renders as intended except for the black level. Either that or there is some sort of configuration for contrast but I haven't seen that.

The main reason I think it's not a problem with the hardware is that it worked perfectly across multiple reboots and only stopped working once I uploaded the example demo code.

stefan-sherwood, 2021-08-23:
The ambient temp during testing is 75° F (24° C)
The light level is medium/dim indoor incandescent lighting

martinberlin, 2021-10-03:
Please try to use the EPDiy component directly and check if it makes a difference
https://github.com/vroland/epdiy

stefan-sherwood, 2021-10-03:
An update on this: the vendor sent me a replacement unit and it works great. I am still curious what would cause the old problem but at least I have a working unit.

TheoKriszt, 2022-02-20:
Sorry to bring this issue back from the dead, but I received a unit with a very similar issue (bought [here](https://www.aliexpress.com/item/1005002006058892.html) back in september 2021).
My first few tries were successful until I suddenly lost quite a bit of contrast on the display without apparent reason. 
I tracked down the issue to the +15V regulator U1 that somehow failed after merely 10 cumulated minutes of use: it was getting quite hot and crashed the +22V line to barely 4V, hence the very degraded contrast.

Anyway, I just wanted to leave a trace here just in case someone else gets the same hardware failure (yep, it's not a software issue). Perhaps the manufacturer could take a look on this issue for the  next board revisions ? 
In the meantime, I just bypassed U1 and lowered the positive output voltage of the LT1945 to around 14V, works like a charm now!

stefan-sherwood, 2022-02-20:
That's very interesting and great detective work. Can you share any more details about how I might look at this on my old non-functioning hardware? Pictures of the relevant components and/or schematics would be fantastic. I'm sure we're not the only ones.

TheoKriszt, 2022-02-21:
Sure!
Take a look at the center-top part of the [schematic](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/schematic/LilyGo-EPD47.pdf) first.
My working hypothesis was that the power supply going to the display section was malfunctionning. The supply consists of a negative(-20V) and positive(+22V) rails provided by the U2 chip, a.k.a. LT1945. 
Take a voltmeter and check these voltages while the board is connected to a power source. 
You can also check wether one of the linear regulators (the U1 and U6 chips) is heating up, even so slighlty.
In my case U1 was the culprit, bringing the LT1945 to its knees.
[image]
After removing U1, the +22V rail recovered but the display had lost its +15V supply. I didn't want to order a similar regulator so instead I got creative and replaced R9 with a 200kΩ resistor to change the +22V in a +14V-ish output. The red wire bypasses the former U1 regulator BTW.

stefan-sherwood, 2022-02-21:
I will take a look. Great stuff.

bigdoug2005, 2023-02-07:
I have two of the LilyGo 4.7 displays running and have noticed a few things that seem to cause the display to fade. One is temperature. I originally did not use sleep and noticed that eventually the whole board would warm up and cause bad fading and banding. I then decided to try to put the device to sleep for 15+ minutes between updates but sometimes the text being displayed will fade to practically fully white. 
My firmware wakes the ESP32 every 15 minutes to check for updates. It seems that the mere act of waking up causes the display to fade. Today I updated the display and unplugged it and the text stayed mostly intact. I just checked all 4 voltage rails and they are fine.

Here is my code for updating and sleeping
```
void epd_update() {
    epd_poweron();
    delay(1000);
    epd_hl_update_screen(&hl, MODE_GC16, temperature);
    delay(1000);
    epd_poweroff();
    delay(1000);
    epd_deinit();
}
```  

followed by
`esp_deep_sleep(1000000*60*15); //15 minutes`

Fresh Update:
[image]
Over Time:
[image]