Xinyuan-LilyGO/LilyGo-EPD47 #58: Compilation error: rmt_ll_enable_tx_end_interrupt (in rmt_pulse.c line 55)

aceisace, 2021-09-06:
Hi and thanks for the awesome software.

I wanted to point out a bug I found while compiling. These are the steps to recreate the bug:
* Use latest ZIP version of this repo and add it via [Add .zip library] on Arduino IDE
* Update ESP32 board package from Board Manager from 1.0.6 (previously used) to 2.0.0
* Try compiling/uploading demo code to ESP32

Compilation does not complete, with the error: `no reference to "rmt_ll_enable_tx_end_interrupt"` (rmt_pulse.c, line 55)

The snippet in question is:
```c++
#if ESP_IDF_VERSION_MAJOR >= 4
    rmt_ll_enable_tx_end_interrupt(&RMT, row_rmt_config.channel, true);
#else
    rmt_set_tx_intr_en(row_rmt_config.channel, true);
#endif
```

olup, 2021-09-09:
Got the same, found a fix ?

aceisace, 2021-09-12:
Not really, but if you downgrade the ESP32 board package to v`1.0.6` again from the board manager, it seems to work fine.

Symbiatch, 2021-09-13:
Open file `rmt_pulse.c` in the library, in the beginning there's commented out lines:

```
// #if ESP_IDF_VERSION_MAJOR >= 4
// #include <hal/rmt_ll.h>
// #endif
```

Uncomment them and code compiles again. The missing function is in that header file.