Xinyuan-LilyGO/LilyGo-EPD47 #59: Only 1 GPIO for users?

Hendrikxs, 2021-09-15:
New user here...
Very happy with the EPD47, but now looking to interface to the outside world.

It seems to me that on connector P5 we have IO12, IO13 and on P4 we have IO14, IO15.
Now the touch panel on P2 uses IO13, IO14 and IO15 leaving only IO12 for users. Is this correct?

Regards,
Henk

cgreening, 2021-09-17:
It certainly looks that way to me - if you only need inputs then potentially you could use the gpios that are connected to the buttons - you'd have to solder some wires onto them...

Hendrikxs, 2021-09-17:
Sounds like a good idea, but the buttons should be disconnected to prevent shorting the output of the other device. 
Maybe hotglueing a connector onto the pcb.

martinberlin, 2021-10-03:
I also think that this is kind of a blocker. But of course esp32 has not so many GPIOs and counting the 8 data bus plus I2C and buttons does not leave much left.
I hope one day Lilygo decides to make a version 2 with ESP32S2 that has more GPIOs and then we can have at least 6 more free IOs

Hendrikxs, 2021-10-03:
"... if you only need inputs then potentially you could use the gpios that are connected to the buttons ...."

I think these GPIO's may serve also as outputs.

AndreKR, 2021-11-12:
The touch screen is I²C, isn't it? So you could connect a port expander like an MCP23017 at the same time as the touch screen.

Yardie-, 2022-02-19:
The easiest way that I can see is to use I2C and connect to an external device that way.
There is lots of i2c stuff and you could add another Arduino that reports over i2c and controls heaps of stuff. 
Or Serial on the USB should work also.

Yardie-, 2022-02-19:
@martinberlin Why would you lose the touch control.
It is just another device on the i2c bus of which the lilygo is a master.

You just use the same i2c Wire connection as the touch controller which is broken out at the pins.
the only thing you would have to be careful of is that the power pins of the output are 5V and the i2c inputs are 3.3 volts.

ie starting the touch library does via the example
```
pinMode(TOUCH_INT, INPUT_PULLUP);
Wire.begin(15, 14);
touch.begin();

```
surely then a simple (as in the basic wire example )
```
Wire.requestFrom(8, 6);    // request 6 bytes from peripheral device #8
while (Wire.available()) { // peripheral may send less than requested
    char c = Wire.read(); // receive a byte as character
    Serial.print(c);         // print the character
  }
```
will talk to your slave
most libraries can take the instance of your Wire class as an argument in the begin method of the class.

martinberlin, 2022-02-19:
you won't loose it, but if touch int pin signals there is an event to read and you are doing another thing it will have to wait until it's possible to receive the data. And touch can be quite data demanding and generates a lot of events if you move the finger around the screen. so I'm not sure if that will work correctly when you are using some other devices there not to mention it's quite hard to get this cables connected without hacking the touch connector itself.

Yardie-, 2022-02-19:
@martinberlin But 12 13 14 and 15 are broken out as the external connectors.
Well they are on my board anyway.
so just hook into those you don't need to touch the touch connector.

Yardie-, 2022-02-20:
I was just looking through the code as this is something I wanted to go properly and 
if you want to stop the touch while doing other things call sleep and wake on the touch object.
or just stop asking for the position on each interrupt.
You could also run the touch routines in another task (fork) with a lower priority.
That is how I plan to do it.
As with screen updates. 
Forking with the esp is really easy and worth looking into.
You can use both cores and set priorities of different tasks.

Yardie-, 2022-02-21:
I WAS WRONG.
unlike the M5epaper 
the lilygo puts out 3.3 volts from the pins next to 14 and 15
So you can hook any i2c device that will run at 3.3 straight up to them.

Yardie-, 2022-02-23:
@martinberlin You suggested that when you connected your devices the touch would not work.
Somethig seems weird with the i2c bus and I am trying to work out what is going on. It seems that whenever another device is connected the touch will not even initialise at all. 
Is that what you meant.

martinberlin, 2022-02-23:
No, I just meant that touch is not another pull data device, and it might be not working 100% correctly if being used along another I2C things that are also being polled in the same port. 
But I meant that only because if you see how it works, generates a lots of events, but theoretically I2C is designed to connect many devices with same 2 wires. But touch is just different because it's like a push request, that's why has an INT pin, and you read the data only when that happens.

Yardie-, 2022-02-23:
Thanks @martinberlin. 
Yeah I thought thats what you were saying the other day.

Obviously I have some other issue. 
It's nice to know someone had this working.
It's me not the LilyGo
Thanks heaps for your time. 

Greg

closed 2022-09-02 by liangyingy