Xinyuan-LilyGO/LilyGo-EPD47 #60: Pullup resistors on the buttons makes them useless for deep sleep

cgreening, 2021-09-18:
Just hit a major roadblock on a project.

The buttons have pull up resistors which makes them pretty much pointless for deep sleep mode.

`esp_sleep_enable_ext1_wakeup` has two modes, `ESP_EXT1_WAKEUP_ANY_HIGH` or `ESP_EXT1_WAKEUP_ALL_LOW`

You can only work out which pin woke the device using the `ESP_EXT1_WAKEUP_ANY_HIGH` - unfortunately this mode is not available as the pins are pulled high. So the device wakes up instantly.

I think the only alternative is to write some ULP assembly.

Kaibin11, 2021-10-06:
Can I try ESP_EXT1_WAKEUP_ALL_LOW?Set only one key and low level wake up

cgreening, 2021-10-07:
I've got some ULP working for this - it's part of my e-Reader project but you should be able to take the relevant bits out. If there's interest I can turn this into a simple example.

https://github.com/atomic14/diy-esp32-epub-reader

closed 2021-10-07 by cgreening