Xinyuan-LilyGO/LilyGo-EPD47 #65: Using other displays

giladaya, 2021-12-20:
Is there a way to make this board work with a 6" display (for example an ED060XC3)?
I know it can work with a specific 5.2" screen (ED052TC2), but it's still not large enough for my needs.

Thanks

LilyGO, 2021-12-21:
The socket on the board seems to be incompatible with ED060XC3, epdiy is adapted, you can refer to this link https://github.com/vroland/epdiy.git

giladaya, 2021-12-22:
Yeah, I know about epdiy, but producing an actual board is not trivial so I was hoping there's a way to use lilygo-epd47 somehow (maybe using a simple cable adapter).
Sorry, I still have a lot to learn about driving EPD's

closed 2022-09-02 by liangyingy

upisqueen-blip, 2026-05-22:
I've reviewed your open-source repo, but it seems everything is tailored for the 50-pin ED052TC4. I'm currently working with the 40-pin ED052TC2 (also on an ESP32-S3). Would it be possible for you to share any reference schematics or code examples to get this specific screen running? It would save me a lot of trial and error. Thanks in advance!