Xinyuan-LilyGO/LilyGo-EPD47 #66: 3V3 and VDD3V3 nets connected?

benkokes, 2022-01-07:
Hello,

I am trying to see where the 3V3 and VDD3V3 nets are connected. Nothing on the schematic suggest a connection between them.

The 3V3 net measures 0.7V when the 18650 battery is used. When connected to USB, the 3V3 net shows 3.3 VDC.

Can you please comment on the nets? I

liangyingy, 2022-01-12:
3V3 is controlled by two networks, VBUS and PWR_EN.

This is for low power consumption.

When you use 18650 battery, you need to call `epd_poweron()`, the 3V3 network will be around 3.3V.

Please see the circuit in this section below:

[image]

pjmi1, 2022-01-13:
Hello liangyingy
Please would you make available to us your schematic of the whole TTGO T5 4.7 (Lilygo-EPD47) circuitry - the diagram above shows that what you have is much more readable compared to LilyGo-EPD47.pdf shown elsewhere?
Thanks

liangyingy, 2022-01-13:
这是来自QQ邮箱的自动回复邮件。
 
你好，我已经收到您的邮件。我会尽快查看并给你回复。

closed 2022-01-18 by liangyingy