Xinyuan-LilyGO/LilyGo-EPD47 #69: 3D-printed STL case issues

scumola, 2022-02-28:
I 3D printed the case that was provided with my resin printer and the 5 physical buttons were bent while putting the EPS47 into the case.  The plastic 'button' part of the buttons was bent upwards and now only click if they're bent back down - I'll probably have to order a whole new display because of this (discount for beta-testing?).  The case needs to be redesigned.  So-far the changes that are needed are:

* Buttons break (plastic 'button' part gets bent and needs to be replaced) when snapped into case
* USB-C connector hole is too high (needs to be closer to 'ground')
* 18650 battery hole is too narrow in both dimensions.  When a battery is inserted into the clips, the clips expand.  Even without a batter in the device, the hole was much too small though.

I've made alterations to the battery hole and the USB-C hole in my own design, but the switches are definitely a problem and mine now have to be replaced.

G6EJD, 2022-02-28:
So you made a design error in the 3D case!
the battery connections are fin if using a standard 18650 cell - i.e. not with inbuilt protection.
The switches are standard pcb types, they are not for industrial/heavy use.

scumola, 2022-02-28:
Perhaps pictures will help.  Note all other dimensions are pretty on-point, so I know that it's not a resolution or accuracy thing with my printer.  Again, I'm using a 2k resin printer, so resolution is pretty good.

[image]
[image]
[image]

I can post my version of the case STL if/when I've perfected it and anyone is interested.

G6EJD, 2022-02-28:
To be honest, I think your still learning how to design cases and it’s not about accuracy or resolution it’s about thinking through the board insertion/extraction process and usage and allowing tolerances for components because you can measure all those from the bare board.

scumola, 2022-02-28:
To be honest, I think that you've never actually wedged the board into your case.  The switches break!  The battery clips don't fit!  You can't design a case for a board just by measuring it alone.  You have to actually test your design.  I've tested the original design and to be honest, it's crap and damages the board.  The resolution/accuracy comments were regarding the 3D printing process, not the design.

closed 2022-09-02 by liangyingy