Xinyuan-LilyGO/LilyGo-EPD47 #75: Can't upload sketch when SD card connected

beanbo, 2022-05-07:
I have this error:
esptool.py v3.1
Serial port COM3
Connecting........_
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:f5:20:55:52:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0008ffff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

beanbo, 2022-05-08:
I found only one solution. It is necessary to take out the module for reading cd cards before loading the sketch. And only after loading you can put the module back.

liangyingy, 2022-05-13:
Upload the log of the sketch when the sd card connected

```
esptool.py v3.3
Serial port COM20
Connecting............
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:cd:ac:7b:6a:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00094fff...
Flash params set to 0x023f
Compressed 18560 bytes to 12759...
Writing at 0x00001000... (100 %)
Wrote 18560 bytes (12759 compressed) at 0x00001000 in 0.4 seconds (effective 342.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 463.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 609.9 kbit/s)...
Hash of data verified.
Compressed 541728 bytes to 343100...
Writing at 0x00010000... (4 %)
Writing at 0x0001e782... (9 %)
Writing at 0x00022875... (14 %)
Writing at 0x000270c2... (19 %)
Writing at 0x0002b1cc... (23 %)
Writing at 0x0002f259... (28 %)
Writing at 0x00033307... (33 %)
Writing at 0x0003733f... (38 %)
Writing at 0x0003b414... (42 %)
Writing at 0x00043960... (47 %)
Writing at 0x00050c5c... (52 %)
Writing at 0x00056255... (57 %)
Writing at 0x0005bcc5... (61 %)
Writing at 0x00061350... (66 %)
Writing at 0x00066952... (71 %)
Writing at 0x0006b9b3... (76 %)
Writing at 0x00071f48... (80 %)
Writing at 0x0007aedf... (85 %)
Writing at 0x000845a7... (90 %)
Writing at 0x00089b2e... (95 %)
Writing at 0x0008ef0e... (100 %)
Wrote 541728 bytes (343100 compressed) at 0x00010000 in 5.3 seconds (effective 811.2 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
```

beanbo, 2022-05-16:
esptool.py v3.1
Serial port COM3
Connecting........_
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:f5:20:55:52:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00118fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

closed 2022-05-16 by beanbo