Xinyuan-LilyGO/LilyGo-EPD47 #78: Cannot Upload with Arduino IDE "A fatal error occurred: Failed to write to target RAM (result was 01070000)"

webfraggle, 2022-06-26:
I cannot upload any example file via the arduino IDE. I got the error "A fatal error occurred: Failed to write to target RAM (result was 01070000)"

My settings are:
[image]

Can someone help me?

G6EJD, 2022-06-26:
Which device are you writing to? Does it actually have PSRAM fitted?

webfraggle, 2022-06-26:
@G6EJD It is a lilygo EPD47 and yes this should have PSRAM onboard

webfraggle, 2022-06-26:
It seems to be a mac problem. I have a m1 mac with Monterey 12.4, the latest serial drivers worked in the past with some other esp32 boards. I hope my old boards will work in future.

I reactivated my windows machine and i was able to upload a bare minimum sketch. So I have to develop on my windows machine for this device.

tavdog, 2022-06-27:
You most likely need the wch serial driver.  Since the chip shortage the CP21xx chips are being replaced with the another version which.  You need another serial driver.  It seems sketchy but it is legit, even Adafruit has had to use them https://learn.adafruit.com/how-to-install-drivers-for-wch-usb-to-serial-chips

capnahab, 2022-08-16:
Tavdog suggestion worked a treatment for me with the same error on a mac (monterey 12.5). Good to see you here David.

tavdog, 2022-08-16:
> It seems to be a mac problem. I have a m1 mac with Monterey 12.4, the latest serial drivers worked in the past with some other esp32 boards. I hope my old boards will work in future.
>
> I reactivated [...]

Did you try the wch driver ? I guarntee it will work.  Time to close this issue ?

closed 2022-09-02 by liangyingy

aaronpk, 2022-11-19:
I'm getting a similar error to this:

```
A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

I'm also using an M1 MacBook Air on Monterey 12.5.1, and I did install the new WCH driver and rebooted the system first.

G6EJD, 2022-11-19:
Wrong board type selected.

tavdog, 2022-11-20:
> I'm getting a similar error to this:
> 
> ```
> A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
> ```
> 
> I'm also using an M1 MacBook Air on Monterey 12.5.1, and I did install the new WCH driver and rebooted the system first.

did you try installing the wch serial drivers ?