Xinyuan-LilyGO/LilyGo-EPD47 #82: fatal error: esp_lcd_panel_io.h: No such file or directory

lordeleren, 2022-09-08:
So I've tried compiling the demo code, but all i get is that there is no esp_lcd_panel_io.h. I've tried removing the include part of it, but this only causes more errors. I cannot find any good places to download esp_lcd_panel_io.h either. 
Does anyone have any idea on how to fix this?.

The whole error message:
"
C:\Users\USER\Documents\Arduino\libraries\LilyGo-EPD47-master\src\i2s_data_bus.c:14:30: fatal error: esp_lcd_panel_io.h: No such file or directory
Multiple libraries were found for "SD.h"
compilation terminated.
 Used: C:\Users\USER\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\SD
 Not used: C:\Users\USER\Documents\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32 Dev Module.
"

liangyingy, 2022-09-09:
[image]

lordeleren, 2022-09-16:
I tried following those instructions, but it didnt work.
But somehow after a week its working fine

closed 2022-09-16 by lordeleren