Xinyuan-LilyGO/LilyGo-EPD47 #84: Only images from example drawimages are working at this moment

DjamesSuhanko, 2022-10-02:
Hi!
I'm Djames Suhanko, blogger of dobitaobyte.com.br.
I would like to know which algorithm  I should use to convert a RGB32 to grayscale 4 bits, as recommended in the sketch drawimage, included in examples directory.
Another doubt related to that is "MSB first"> What this means? Do I need to shift right the most significant bits?
To finish the main points, I saw in pic2.h (used as example here) the bytes values are very big to be for bits:
`pic2_data[(960*576)/2] = {
0xAA,...`
I did  a program to convert RGB888 to RGB565, grayscale 8bits and monochrome and I will do articles and videos using this display to exemplify the grayscale usage. To exemplify monochrome I will use the TTGO and LilyGO T-Display. The program is free, I need only finish this conversion to share with all.

My program is made in Qt and I tried two types of grayscale, 8 and 16. So, I decided to write a loop to access pixels directly and my choice to do the conversion was luminance:
`QRgb target = this->localImage.pixel(i, j);
QRgb lum = 0.2126f * qRed(target) + 0.7152f * qGreen(target) + 0.0722f * qBlue(target);`
After that, I shifted 4 bits to right:
`lum = (lum >> 4)`
This way, I have only 16 values. Ok, so, I write this value back to image, and now it's no more a RGB, as expected. Resuming: My Image is a array of a single byte for pixel.
Where is my mistake? Could you help me to solve that, please?

asozio, 2022-10-02:
Not 100% sure, but noticed you mentioned grayscale 4 bits (nibble), but your lum value is 16 values (single byte)
Before writing the file, wouldn't you need to either convert your single byte lum into nibbles (4 bits) or merge two lum values (each 4 bits) into a single byte?

DjamesSuhanko, 2022-10-02:
Wow, I thank you for answer, asozio. It's a light to me, because I don't have idea how to convert it correctly. 
I will start read the link right now and back here to say what happened after try it.

DjamesSuhanko, 2022-10-03:
Yupi! It's working! Thank you, @asozio !
My image is flipped but I will take a look better now. I didn't know about name "nibbles". Thank you for share this important knowledge with me!!

closed 2022-10-03 by DjamesSuhanko

asozio, 2022-10-03:
> Yupi! It's working! Thank you, @asozio ! My image is flipped but I will take a look better now. I didn't know about name "nibbles". Thank you for share this important knowledge with me!!

Glad I could help!

G6EJD, 2022-10-03:
MSB first means Bigenden that means literally that send the MSB first, then next MSB and finally LSB

DjamesSuhanko, 2022-10-05:
My program is working now, but I need to refactor interface and puts the thing in classes. Until version 1.0 everything will gets fine.
Now any maker can create an image array to show on display T5! \o/
https://github.com/DjamesSuhanko/Image2Display