Xinyuan-LilyGO/LilyGo-EPD47 #87: [ESP32-S3] Battery ADC with wi-fi?

nickolay, 2022-11-29:
The ESP32-S3 based "T5-4.7 Plus" provides IO14 for reading the battery voltage. In terms of ADC, [it is](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf) known as `ADC2_CH3`, and ADC2 is [said](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/api-reference/peripherals/adc.html#adc-limitations) to be conflicting with wifi operation. Common wisdom on the Internet is to avoid attempting to use channel 2 ADC together with Wi-fi -- I couldn't even find any success reports. The battery examples in this repo are, suspiciously, running with wifi off, and the wifi example doesn't monitor the battery.

Before I spend too much time on this... Did anyone get this working? Could you provide an example?

lbuque, 2022-12-12:
Yes, ADC2 and wifi cannot work at the same time. This is our mistake.

nickolay, 2022-12-12:
Thanks for confirming my findings! Do you know if there's any workaround worth pursuing? Or is it only possible to read battery level after booting the device and before the wi-fi is turned on for the first time?

lbuque, 2022-12-12:
You can turn off the wifi (esp_wifi_stop) before reading the adc value.

nickolay, 2022-12-12:
Good, I'll try that sometime -- thanks again!

nickolay, 2022-12-14:
Your example seems to work -- in that it reports decreasing voltages (starting with ~ 4.1+V) while on battery, and 4.6V when no battery is connected. (*)

I am confused about the accuracy of these measurements though, given that:
* I'm not sure what negative effect wi-fi was to have on ADC2 (the docs [used to say](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#adc-calibration) the reading would "fail" -- could I be seeing results distorted by wifi operation? Should I expect occasional random values?)
* The other demo [advises to use `epd_poweron()`, followed by a short delay](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/deae2d2cbfb062198e6b4647d29eab6bc4d80d8a/examples/demo/demo.ino#L133-L136), while the code here just reads the ADC without all the ceremony
* I don't seem to have the ADC calibration values (though, judging by [this graph](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#adc-calibration), the effect should be minor)

I suppose I can just see how it goes, unless you have any additional advice.

Your help so far is very much appreciated. Thank you again!

(*) I didn't know how to view `Serial` output while not connected via USB, so I had to add `log_wifi(String(battery_voltage));`, which is defined like below.

```
void log_wifi(String s) {
    WiFiClient client;
    const int httpPort = 8888;
    String host = "192.168.2.88";
    if (!client.connect(host.c_str(), httpPort)) {
        Serial.println("connection failed");
        return;
    }

    String url = "/?v=" + s;
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }
}
```

closed 2023-01-03 by lbuque

random-0110-dude, 2023-04-22:
@lbuque Thank you for this example. It works, but it often triggers watchdog and thus a reboot happens. Can it be solved as well?

berry0511, 2024-05-24:
Hello everyone, I would like to ask, if I enable the WiFi function on the ESP32-S3, will it only affect the ADC function, and will the functionality of GPIO/SPI/I2C be affected?