Xinyuan-LilyGO/LilyGo-EPD47 #88: "waiting for download" after manual Reset

BatsIhor, 2022-12-02:
Hi, 

Quite often when I press Reset button device stuck on "waiting for download". It also happens when waking up from sleep which in turn drains battery and renders it useless to client. 
Normally after Reset I see: 

16:24:23.487 > rst:0x1 (POWERON_RESET),**boot:0x13** (SPI_FAST_FLASH_BOOT)
16:24:23.487 > configsip: 0, SPIWP:0xee
16:24:23.487 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:24:23.487 > mode:DIO, clock div:2

But often I get this message and nothing works:

16:24:25.292 > rst:0x1 (POWERON_RESET),**boot:0x1** (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))
16:24:25.292 > waiting for download

On the espressif forum there is a topic related to this issue and looks like it's might be solved by adding additional capacitor. 
Is it possible to add it manually or implement software fix? 

Thank you
Ihor

lbuque, 2022-12-12:
The boot button is not allowed to add capacitors.

BatsIhor, 2022-12-12:
On the other boards usually you have to connect IO0 to GND to boot in download mode, but with this board it just randomly happens and I have 30 of them that show the same behavior

lbuque, 2022-12-13:
Whether you touch the boot button by mistake, Espressif's reference design does not add capacitors. Adding a capacitor instead makes it easier for the esp32 to enter download mode.

BatsIhor, 2022-12-16:
Just to make sure we are on the same page:
"waiting for download" happens randomly after reset, and I'm not touching any other buttons than RESET. 
I'm looking for a way to make sure it doesn't happen, as this prevents my code from execution.

brianwyld, 2023-08-18:
If P0 (STR_IO0) reads low at reset time, the ESP32 enters the DOWNLOAD_BOOT mode.

On the Lilygo T5-4.7-S3 (version using ESP32-S3, with 3 buttons) there is a pull up resistor on this button to ensure it is high if not pressed.

On the T5-4.7 (ie previous version with ESP32, with 5 buttons), the STR_IO0 input does not have a pullup resistor (on the schematic at least) so will float -> value at reset could be anything... 

So I suspect you need a pullup _resistor_ (say 4.7kohms) to make it work reliably.

btw, I note that the other buttons have a capacitor in parallel with the button, but not STR_IO0. I assume this is to debounce the switch a bit - but made STR_IO0 be low at power up time so broke the boot process, so they didn't put one on this signal?

I have the S3 version, and haven't had the problem you see (but I have othe problems... eg although I can read the button value correctly,  I cannot get an interrupt when the pressed or released....)

closed 2024-12-07 by github-actions[bot]