Xinyuan-LilyGO/LilyGo-EPD47 #89: Problem loading weather project to t5-4.7 S3 board

Jxid, 2023-01-25:
Hi i have problem loading weather project to t5-4.7 S3 board.
I starts uploading and at 80% of uploading i get a disconnect error and upload fails.
I have tried to upload demo sketch and it worked fine.
I am using Arduino IDE 2.0.3

G6EJD, 2023-01-25:
It will not work.

Jxid, 2023-01-26:
Why?

G6EJD, 2023-01-26:
Piracy of code

Jxid, 2023-01-26:
OK Thank you. Is there any other project for testing my board?

lbuque, 2023-01-30:
Did you upload this [code](https://github.com/Xinyuan-LilyGO/LilyGo-EPD-4-7-OWM-Weather-Display/tree/web)?

G6EJD, 2023-01-30:
No, that’s the issue, it’s a blatant copy of my code.

lbuque, 2023-01-30:
Yes, this code is from https://github.com/markbirss/LilyGo-EPD-4-7-OWM-Weather-Display.

I learned from the leader that this code has been allowed by the developer for us to use. If you take issue with this copy, we may delete this copy.

G6EJD, 2023-01-30:
I am the developer and I didn’t allow it. In that link the individual removed substantial parts of my licence and assumed that was ok, which is not.
Please remove my code.

closed 2024-12-05 by github-actions[bot]