Xinyuan-LilyGO/LilyGo-EPD47 #91: Draw an image from the SD card

FrancescSobrevela, 2023-02-22:
Hello again.
In my project now I need to draw an image on the screen. This image is inside the SD card. Anyone knows how can I do it? 
Thanks

lbuque, 2023-02-24:
@FrancescSobrevela Can you refer to the following code, is it helpful?
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/examples/wifi_sync/wifi_sync.ino#L418-L432

Zeyin, 2023-08-21:
Thanks @lbuque!

But I got this error while following the link provided. Do you know what might be the root cause of the below stacktrace?

09:44:39.151 -> Try to display img 2.jpg
09:44:41.064 -> Image length is 148463
09:44:41.064 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
09:44:41.064 -> 
09:44:41.064 -> Core  1 register dump:
09:44:41.064 -> PC      : 0x400570e8  PS      : 0x00060f30  A0      : 0x8200be7a  A1      : 0x3fcebd10  
09:44:41.064 -> A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0007e900  A5      : 0x00000000  
09:44:41.064 -> A6      : 0x00000016  A7      : 0x00007e90  A8      : 0x8202d57d  A9      : 0x3fcebd80  
09:44:41.064 -> A10     : 0x3fcebe2c  A11     : 0x3c0fa708  A12     : 0x00000016  A13     : 0x00000100  
09:44:41.096 -> A14     : 0x3fca0a40  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
09:44:41.096 -> EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00007e8f  
09:44:41.096 -> 
09:44:41.096 -> 
09:44:41.096 -> Backtrace: 0x400570e5:0x3fcebd10 |<-CORRUPTED
09:44:41.096 ->

ZaleAnderson, 2023-11-09:
I have an img on my SD Card but every time I run the wifi_sync program it just displays that the FatFS initialization failed with no other output. This is the output from the websever 

`FILE SYSTEM Not Detected

URI: /edit
Method: GET
Arguments: 0`

Any help would be appreciated. I cannot get this sketch to work.

clajarac, 2024-05-15:
same problem, help please XD

ZaleAnderson, 2024-05-16:
I ended up skiping the example code to take an image from the sd card and instead wrote it to the buffer using the FS, SD, and SPI library (correct me if I am wrong, it's been a long time since I looked at this code)

`uint8_t *framebuffer;
#define FILE_SYSTEM SD
static bool hasFILE_SYSTEM = false;`

`void setup() {
    SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
    bool rlst = SD.begin(SD_CS, SPI);
    framebuffer = (uint8_t *)heap_caps_malloc(EPD_WIDTH * EPD_HEIGHT / 2, MALLOC_CAP_SPIRAM);
    memset(framebuffer, 0xFF, EPD_WIDTH * EPD_HEIGHT / 2);
}`

`File jpg = FILE_SYSTEM.open("PATH TO IMAGE ON SD");
  String jpg_p;
  while (jpg.available()) {
    jpg_p += jpg.readString();
  }
  Rect_t rect = {
    .x = 330,
    .y = 39,
    .width = EPD_WIDTH,
    .height = EPD_HEIGHT,
  };
  show_jpg_from_buff((uint8_t *)jpg_p.c_str(), jpg_p.length(), rect);`
  
  However it does seem some images cause a freakout and crash the esp. I do not know what causes this, it could be from me converting them to black and white right on the esp and it doing some bad math.

closed 2024-12-04 by github-actions[bot]