Xinyuan-LilyGO/LilyGo-EPD47 #93: T5-4.7 Plus S3 hardware issues & partial clear corruption

AdriVerhoef, 2023-03-26:
Hello all, I got my T5-4.7" S3 a week ago and it's a bit of a let down. Straight out of the box I notice a few lines of vertical dead pixels, ghosting on writing text and each time the screen refreshed the PCB made a high pitched noise. The repair demo did not fix the dead pixels. For the noise I expected to swap some capacitors to fix the issue, but it was not that easy.

I probed the +22V and -22V outputs and noticed high ripple (~3.5Vpp) on the +22V each time the screen updated:
[image]

Changing the output capacitors for X7R 50V 1uF reduced this a bit, but not much. Adding additional capacitance wasn't helping either. On further probing I discovered that the +15V output of the LM7815 was oscillating. Adding more capacitance (2x22uF 25V X5R) made things better, but the output still had high ripple.

In the schematic I found that the second op-amp in the dual op-amp package was not terminated properly. The second opamp's inputs where left floating and most likely oscillating:
[image]
I soldered the opamp inverting input and the output together and added a wire to the first op-amp's non-inverting input, so the op-amp was in unity gain configuration, just like the first opamp. I did not connect both op-amp outputs as this is not a good idea, since both op-amps can have a different offset.
[image]

With the op-amp properly terminated and additional capacitance on the +15V output the display was now silent on each update and text printing on wrong locations was eliminated. 
I also resoldered the pins of the ESP32S3, as I had to use a microscope to confirm that there was at least some solder on some pads. This did not improve functionality, so the pads where properly connected before resoldering even though it was discouraging how little solder was on some pads:
[image]

On the diy-epd support I also found a comment that the VCOM should be buffered with an additional capacitor for improved performance, so I added a 22uF capacitor with a 10 Ohm series resistance in the output trace of the opamp output. I first scraped the soldermask from the output trace, then cut it, so the 10 Ohm is in series to the display VCOM input and soldered the 22uF capacitor to the resistor and soldered it to the ground plane. Most op-amps do not like driving a high capacitance on the output, so the resistor is used to isolate the capacitance from the output and keep the output stable. Though this modification seems to have the least impact on the functionality of the display.

All modifications:
[image]
The 47uF elco is not necessary, but is a left-over from testing if additional capacitance solved the issues left.

I still have the issue that the display shows a block around the parameter which is updated due to the rest of the display slowly getting corrupted/darkening when doing a partial clear:

I've changed my VCOM voltage by adjusting the potentiometer to ~-1.1V which reduced the corruption significantly, but still with each area clear the previously displayed data corrupts more and more over time, while the refreshed data is clear.

The epd_push_pixels function does not seem to write a complete block of pixels. The last row being cleared does not seem to write a 1 or 0 to the screen, but most likely something random in a buffer.
[image]
This causes some residue to remain after clearing:
[image]

So I hope someone has a solution for the partial clears corrupting the full screen as I suspect it's a software problem. Erasing and rewriting the full screen using the framebuffer is a solution, but I hope it's not required doing so after a single parameter changes.

Here is a code snippet for a printf like function in order to make the examples a bit more clean:
```
#define TEXT_BUFF_SIZE 128
void printLine(int32_t x, int32_t y, const char * format, ... )
{
    static char local_buff[TEXT_BUFF_SIZE] = {0};
    int32_t cursor_x = 0;
    int32_t cursor_y = 0;
    va_list args;

    // epd_poweron();
    memset(local_buff, 0, TEXT_BUFF_SIZE);
    va_start (args, format);
    vsnprintf(local_buff, TEXT_BUFF_SIZE, format, args);

    cursor_x = x;
    cursor_y = y + FiraSans.advance_y + FiraSans.descender;
    writeln((GFXfont *)&FiraSans, local_buff, &cursor_x, &cursor_y, NULL);
    va_end (args);
    // epd_poweroff();
}
```
Usage:
```
    epd_poweron();

    printLine(200, 250, "➸ 16 color grayscale  😀 \n");
    delay(500);
    printLine(200, 300, "➸ Use with 4.7\" EPDs 😍 \n");
    delay(500);
    printLine(200, 350, "➸ High-quality font rendering ✎🙋");
    delay(500);
    printLine(200, 400, "➸ ~630ms for full frame draw 🚀\n");
    delay(500);

    epd_poweroff();
```

random-0110-dude, 2023-04-10:
@AdriVerhoef Thank you very much for sharing!

Could you please provide more close-up photos, what exactly did you solder to silence the board?

AdriVerhoef, 2023-04-11:
[image]
[image]

Mods:
- Orange: 2x 22uF + 1x47uF, but as stated earlier, that didn't improve the ripple voltage as the oscillation was already resolved.
- Red: The connections of the opamps as indicated in the first post to create a voltage follower/unity gain configuration. Added a 100nF to buffer the VCOM voltage from the potentiometer.
- Calibrated the VCOM voltage to 1.1V
- Dark Green: Added an additional 1uF to the positive output of the symmetrical boost converter
- Purple: Changed the pull-ups on the enable from 100k to 10k, since the voltage on those pins was not stable due to the high resistance of the original resistors and the input current of the inputs.

[image]
[image]
Mods:
Cut the trace at the green line and scrape the solder mask from the trace, so the resistor can be soldered. For the buffer capacitor the ground plane needs to be solderable, so remove the soldermask at an appropriate distance from the trace.

So do you have similarly problems with the board? Noise and ghosting? 

I've also discovered that the STR_IO0 signal is connected to two pins on the ESP32S3. Would have been nice is the STR and IO0 functionality would have been seperated, so the shift register and pushbutton where not connected. However the connecting trace seems to be under the ESP32S3 module or on the otherside of the board below the adhesive.

[image]

random-0110-dude, 2023-04-11:
@AdriVerhoef Thanks!

> So do you have similarly problems with the board? Noise 

Noise. Especially when updating the screen. Gone only after `epd_poweroff_all();`.

I did some modifications already, before your answer. Here they are:

[image]

[image]

- I connected 2 with 5 on U5 / LM358. Noise goes a bit down.
- I connected 7 with 6 on U5 / LM358. Noise goes a bit down
- I added 2x22pF ceramic capacitors, because they were lying around. Noise goes a bit more down .
- I finally added 47uF electrolytic capacitor (lying around as well). Noise gone completely.

By the way, your images show connection between 3 and 5, **is this intentional? Or did I mix things up?**

I connected all these without any idea what I'm doing (except maybe that I'm filtering some pulsations). No idea about opamps as well. Noise gone.

**Are my modifications safe for the screen?**

I don't own an oscilloscope, but I measured a couple of voltages, here they are:
[image]

AdriVerhoef, 2023-04-11:
I see I've connected pin 2 in my example of the schematics and pin 3 on my board. It doesn't really matter, the second opamp is now in series with the first, whereas on my board both opamp inputs are connected to the potentiometer.

Just ensure that the leads of your capacitors do not make contact with the rest of the circuit and you should be fine. The 22pF don't do much, but won't harm either. The 47uF is 47000000pF so that adds the missing capacitance the +15V regulator requires, as it is oscillating when the screen is updating.
You do need to reduce the VCOM voltage by turning the potentiometer, the voltage should be around 1.1V-1.3V. When this is reduced, the screen corruption during a partial clear and draw is significantly reduced. However I've programmed my board with the weather example which updates the screen fully, as I still found it too visible.

random-0110-dude, 2023-04-11:
@AdriVerhoef Thanks!

Yep, I double-checked that there are no solder traces left, and the capacitors remain isolated. 

> You do need to reduce the VCOM voltage by turning the potentiometer, the voltage should be around -1.1V ~ -1.3V.

Is VCOM a voltage between 3 and GND? _Sorry for the dumb question_

AdriVerhoef, 2023-04-11:
VCOM is the common voltage of the Eink display, the potentiometer is connected to R10, which is connected to -15V. I do not know the resistance of the potentiometer(U4 in the schematic), but it creates a voltage divider, so the output voltage can be adjusted from 0V(GND) and a negative value. The opamp buffers this voltage as the screen will draw more current than a voltage divider can supply without being affected.
[image]
[image]

If you use a screwdriver and turn the potentiometer slightly, the voltage you measure on pins 2 and 5 will change. Just make sure that the display's power converters are on or the measured voltage will not be correct.

random-0110-dude, 2023-05-01:
@AdriVerhoef And one more question - did you find any issues with RTC?

Mine resets to zeros every time I reboot the board. Is it intended? Does your work the same?

random-0110-dude, 2023-05-03:
@AdriVerhoef The battery shows 2.66V between its PCB pins. 

As far as I know, I don't reset the RTC intentionally. I've skimmed through https://github.com/lewisxhe/PCF8563_Library/tree/master/src , but did not find any code that resets it during initialization.

homonto, 2023-06-13:
I've just bought this board with the display and EVERYTHING you describe is completely true in my case as well (S3 version): the noise, the partial grayout etc.
I think that goes without saying: "you pretend to pay me - I pretend to be a good board" - eh...

btw when you power this board using the BAT connector, when. you enter epd_poweroff_all() it still draws about 270uA that is way too much - if I am going to use this board I would add some circuit to disconnect the power completely and only turn ON periodically (i.e. every 3min or so)

on top of that, the library provided by Lilygo is full of errors - I am wondering if other libraries would work, i.e. TFT_eSPI - that would be amazing

brianwyld, 2023-08-22:
Some comments after using the board for a bit (under micropython):

- the button STR_IO0 is essentially not usable as it is used for the epaper control signals (via the shift reg) -> if you try to use it as a general button then the epaper updates don't work... I'm thinking about cutting the trace and wiring the button to a different IO to be able to use it for the application
- bizarrely I can't get the button IO inputs to generate an IRQ (using micropython, but the code works on other ESP32S3 boards...)
- 50% of the the time when I start the epaper driver, the board browns out (powered by USB) and resets -> I need to try some of your 'add caps' fixes
- epaper driver code is somewhat buggy, specially for partial updates -> need to fix that code for the last line writiing
- power consumption : not fab, I suspect because the buck-boost for the epaper has no way to control its 'shutdown' input
- for the micropython command line, I ended up soldering a header to the GND/RX/TX on the 40 way header to access the UART, instead of using the native USB uart -> doesn't disconnect from the PC at every reset, and lets you see the reboot logs (this is how I saw the brownout reset for example)

definitely feels like a friday afternoon design...

lewisxhe, 2024-11-21:
@Tasshack Thank you for your contribution. I will test it according to your method. If it works well, I may update it according to this hardware connection later.

random-0110-dude, 2024-11-21:
@lewisxhe If you will do a new revision of the board, could you please also look into #98 as well? 
Currently I have to do two modifications:
* https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/93#issuecomment-1503806892 to silence the board
* https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/98#issuecomment-1715584471 to prevent brownout on boot and strange battery behavior: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/98#issuecomment-2267149084

lewisxhe, 2024-11-22:
Subsequent revisions may be based on the model in the picture, and the power supply may be replaced. I will update the progress here later.
[image]

random-0110-dude, 2024-11-22:
@lewisxhe looks really nice! Does it mean that it can be used with any magsafe charger?

Some more items for wishlist:
* please move those two molex connectors a bit apart (up to 5-7mm) from each other to make room for cables
* please add BOARD_I2C connector (always powered, unlike molex ones that require `epd_poweron()`) - for extra versatility
* please consider adding APDS9960 sensor
* please add more buttons, with a capability to wake the ESP. 5 buttons total (BOOT+RST+3 custom) will be enough IMO
* fix those pullups on USB-C (#116) (if not already)

lewisxhe, 2024-11-25:
@random-0110-dude Thank you for your suggestion. Please refer to the official website for the design of this model, because it has been designed and I cannot influence it. I can only improve it according to your feedback in the subsequent design.

random-0110-dude, 2024-11-27:
@lewisxhe well, my suggestions are also valid for the current (V2.4) model. Are you going to sell it alongside the new 'MagSafe' one?

lewisxhe, 2024-11-27:
@random-0110-dude  Currently V2.4 and T5-Epapaer-Pro are of the same design. We will listen to relevant opinions and then summarize and make changes.

lewisxhe, 2024-11-27:
> @lewisxhe looks really nice! Does it mean that it can be used with any magsafe charger?
> 
> Some more items for wishlist:
> 
> * please move those two molex connectors a bit apart (up to 5-7mm) from each other to make room for cables
> * please add BOARD_I2C connector (always powered, unlike molex ones that require `epd_poweron()`) - for extra versatility
> * please consider adding APDS9960 sensor
> * please add more buttons, with a capability to wake the ESP. 5 buttons total (BOOT+RST+3 custom) will be enough IMO
> * fix those pullups on USB-C ([usb-c pull down on B5 missing #116](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/116)) (if not already)

High-speed wireless charging cannot be used. Long-term use will cause severe heat and damage. Only standard wireless charging can be used.
The wish list he mentioned may require another revised version without a shell to achieve the functions he wants. The Pro version currently being made does not meet its needs.
1. There is only one Qwiic interface on the new version. If more are needed, a replacement lora function module on the back needs to be designed.
2. Same as above.
3. APDS9960 is a sensor that requires a separate reserved design for the placement position. There is no reservation on the screen glass of the new version, and it can only be expanded through the Qwiic interface.
4. There are only 2 customizable (GPIO0, GPIO48) buttons on the new version, two function buttons RST and PWR buttons. No more reservations.
5. Fixed.

The above is the reply of Design T5-Epaper-Pro Design Yes.

martinberlin, 2024-12-01:
I’m sorry about voting down to support Lilygo S3 efforts into epdiy. In fact they did got the S3 working first. But the fact that they still use kind of epdiy v2 without really making the efforts to use S3 LCD full signals to drive the whole hardware really put me down.
Please check my analysis here:
https://youtu.be/r4-dz_x7K4k?si=8xecWLySTkGISP_Z

The fact that they’ve used a really BAD quality display in S3 compared to their first WRover version also didn’t helped. I’m a fan of their hardware in other matters but really Lilygo in this particular PCB design you failed badly.
Instead of checking epdiy latest innovations and try to make a compatible PCB hardware you just make this failed version that really serves no one to use in a professional design. I tried to make that video and make my followers understand why I was not going to support this board in epdiy but I was not really been heard

lewisxhe, 2024-12-03:
Hello everyone, we have discussed and decided to directly use the epdiy v7 hardware design, the same driver method and GPIO. What do you think about this?

random-0110-dude, 2024-12-03:
@lewisxhe I think this is a great idea that will allow maximum compatibility and will probably boost your sales :) A **reliable** and **inexpensive** board will definitely be a hit.

In the same time, I personally would not like to see CH340, and would prefer having direct USB access to ESP32-S3, so the board would not lose functionality (USB-MSC, USB-HID etc) that would happen if you would expose only UART over USB. [Other boards](https://www.tindie.com/products/fasani/epdiy-v7-816-bit-parallel-epaper-controller/) have two USB-C ports; I'm not sure if I want two, maybe one is enough, but it should be the one with direct ESP access.

RTC is a must. microSD card slot - also very much so.

Please add more buttons! Five (including RST & BOOT). Also, **all** these buttons should be able to wake ESP from deep sleep.

Please expose both I2C lines available on the ESP. One line (probably the one with RTC+touch) should always be powered, while the second line should be controlled like in v7. **Both** I2C lines should have pin headers available (or at least holes for them).

Probably you could keep ['your'](https://aliexpress.com/item/4000473537275.html) sockets (to keep compatibility) but please expose the same lines via 2.54" pin headers as well.

lewisxhe, 2024-12-03:
1. No USB2TTL or any serial port bridge will be used, but the USB of ESP32S3 will be used directly as the download and debug interface

2. RTC and MICROSD are reserved

3. Five buttons need to be determined based on the subsequent design, because a LoRa module needs to be compatible

4. Public I2C is a must, and it is always powered

5. The second I2C depends on whether it can be enabled, because LoRa is required. I will follow up on the hardware design side. After the schematic diagram is modified, I will send it to you for review.
Perhaps you can switch freely between LoRa or I2C, which is a better idea, because not everyone needs LoRa.

G6EJD, 2024-12-03:
There are no dedicated I2C pins on the ESP32 every pin below 33 can be programmed as an I2C and the whole idea is it's a bus, so no need for a separate port...

lewisxhe, 2024-12-05:
Ladies and Gentlemen, this is the schematic diagram of the modified version according to the design of EPDIY V7, retaining the original GPIO, please check it out. Comments are welcome.

[T-EPD47-PRO V1.0.pdf](https://github.com/user-attachments/files/18019780/T-EPD47-PRO.V1.0.pdf)

martinberlin, 2024-12-05:
This looks so much better @lewisxhe  I will find sometime to review it in the next days but so far I see all control lines there:

```C
static lcd_bus_config_t lcd_config = {
    .clock = CKH,
    .ckv = CKV,
    .leh = LEH,
    .start_pulse = STH,
    .stv = STV,
    .data[0] = D0,
    .data[1] = D1,
    .data[2] = D2,
    .data[3] = D3,
    .data[4] = D4,
    .data[5] = D5,
    .data[6] = D6,
    .data[7] = D7,
    .data[8] = D8,
}
```

[image]

I guess it will be fairly easy to create  a new board in epdiy and add support to this one. I can do it will send you an email later. Thanks for the update

lewisxhe, 2024-12-06:
@martinberlin   Well, any suggestions you may have, we are listening.

martinberlin, 2024-12-06:
You can try it already. Since it's very similar to v7 but with just 8 data channels it should work out of the box (Unless I miss something)
https://github.com/vroland/epdiy/pull/383
Just clone epdiy, switch to branch:   [new/lilygo-s3](https://github.com/vroland/epdiy/tree/new/lilygo-s3)

And there I left an example:
https://github.com/vroland/epdiy/tree/new/lilygo-s3/examples/lilygo_s3
That should render the Dragon demo in your display.  Just tell if it does something. For me it builds without errors but I cannot test it since I don't have the hardware.

> Well, any suggestions you may have, we are listening.

So far the PCB looks very nice and the fact that has LoRa and should be LVGL capable if using touch, can make it a very nice fun toy for a slow chat :P  
Honestly what I didn't like in your first S3 versions is that the display was flexible and for me, it had some dead rows, I've got 2 of them and had the same result (Maybe a bad batch). But I guess by now you would have that sorted out since it was not only me who had this issue. 
Also now the problem with the sound that the PCB generated in initial versions should be gone with a professional dedicated PMIC to generate the voltages. It's also fully compatible with v7 since uses same IO expander and the TPS65185. This is cool hardware to play with but of course makes the PCB costs go a bit higher.

lewisxhe, 2024-12-07:
@martinberlin Thank you. We are just discussing it now, and there is no hardware to test it.

Thank you for your reminder. I will forward it to the hardware engineer and let him see your proposal.

We will send you samples after we have preliminary samples.

lewisxhe, 2024-12-07:
Have you removed all external sensors and then measured the voltage?

This board has only one voltage regulator, which is RT9013 from the schematic diagram. It can only provide 500mA current at most, which may not be enough for ESP32. The maximum peak current of normal ESP32 is generally around 700mA.

random-0110-dude, 2025-01-07:
@lewisxhe I'm also happy with the design. But still I'd like to see more buttons, and wired in a way that it would be possible to wake the ESP from deep sleep. 

Is it possible to add them?

closed 2025-03-21 by github-actions[bot]