Xinyuan-LilyGO/LilyGo-EPD47 #97: S3 is not accessible after going into deep sleep

BatsIhor, 2023-06-06:
Hi there, I have uploaded my sketch with deep sleep after showing a text and now I can't upload anything else as right after it displays image it goes into deep sleep.
I tried to press reset "on time" so I'm able to update code but hope there are other options to make sure I can upload without trying 10 times. 

Can I put it into flash mode somehow? 

Thanks

homonto, 2023-06-13:
if you use "lilygo-t5-47-plus" with S3 in Arduino, you need to set up "CDC Enabled" in Arduino.
If you use platformio you need to use:
```
build_flags =
    -D BOARD_HAS_PSRAM
    -D CONFIG_IDF_TARGET_ESP32S3=1
    -D CORE_DEBUG_LEVEL=0
    -D ARDUINO_USB_MODE=1
    -D ARDUINO_USB_CDC_ON_BOOT=1 
```

I had the same problem and I sorted it out.

But more important: if you send it to sleep immediately after it did the job, your serial monitor might NOT have enough time to realise that the S3 is connected - try to add i.e. delay(10000); before it goes to sleep and check then - later you can remove of course
see below - when it goes to sleep you don't see it at all as power is off
```

[setup] poweroff done: 3796ms
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbmodem2101 .....       Connected!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

pabloazurduy, 2024-09-28:
thanks for the help, what worked for me was to disconnect and connect very fast, during the initialization the port is still recognizable. 
During that small time window, I uploaded a different example.

closed 2024-12-02 by github-actions[bot]