Xinyuan-LilyGO/LilyGo-EPD47 #98: Drop voltage on 3.3V rail after enabling prw_en pin

viteka32, 2023-08-07:
We using and testing LilyGo 4,7 Plus and have a problem with resetting ESP32 after enabling display with sequences epd_init and epd_poweron. We found a bug on PCB. Enabling display generating 0.9A peak and voltage level on LDO 3.3V drop to 2.2V. There is ESP32 randomly resetting. Used LDO doesn't have the capacity for this current peak. I am testing adding a blocking capacitor on 3.3V rail and 1000uF value can be blocking this peak when the voltage doesn't drop under 3V. 

Testing code:
```
void setup() {
  delay(500);
  epd_init();
  delay(500);
  epd_poweroff_all();
  esp_sleep_enable_timer_wakeup(2 * 1000000ULL);  //sleep in microseconds
  esp_deep_sleep_start();
}
```

Current measured:
[image]
The first sequence is correct and the next is wrong. 

The voltage on 3.3V supply during a current surge:
[image]

Have you encountered this problem too? Do you have any solution for this?

brianwyld, 2023-09-07:
Same problem on T5+. I note the original T5 does not seem to have this issue.

Did the 1000uF fix it?

viteka32, 2023-09-07:
Yes, the 1000uF capacitor will prevent the power supply from dropping below 3V. Although this is a fairly short peak, the restart appears to be random. Instead of a capacitor, adding a delay (20 ms) right after switching on the display power branch also helped us a lot.

homonto, 2023-09-07:
1mF capacitor for such small project is totally wrong design. It might be a solution but for sure it is a mistake of the design. Btw where does this 0.89A go?

viteka32, 2023-09-07:
Yes, for this type of design, a 1mF capacitor is not the right solution. For example, with LTE 2G, it would mostly be necessary. From the measurements, we found that the pulse is generated when the DC/DC converter LT1945 and display are switched on. (The greatest credit goes to the LT1945 power supply circuit with accompanying regulators)

homonto, 2023-09-07:
> Yes, for this type of design, a 1mF capacitor is not the right solution. For example, with LTE 2G, it would mostly be necessary. From the measurements, we found that the pulse is generated when the [...]

from the datasheet of LT1945 it has a limit to 350mA however, the peak current can reach easily even 1A - after 0.1ms the regulator is OFF completely.
But, the LT1945 is not a load so this current is NOT related to LT1945 being ON, but to other element, that consumes that current during the transient.
The decoupling capacitors are exactly to reduce the voltage swings, but 1mF is not in the normal range of such capacitors

brianwyld, 2023-09-07:
after reviewing the schematic for the T5-PLUS board, and looking at the driver code, I am confused. 

1/ epaper power supply control:
In the code (ed047tc1.c), I find epd_power_on() and epd_power_off().

For power on, it sets bits named "scan_direction=true", "power_disable=false", "neg_power_enable=true", "pos_power_enable=true" and "ep_stv=true", in order, with a delay between them.

For power_off, its "pos_power_enable=false", "neg_power_enable=false", "power_disable=true", "ep_stv=false", again in order iwth a delay.

HOWEVER:
 - on the schematic, power_disable->SMPS_CTRL, neg_power_enable->NEG_CTRL, pos_power_enable->POS_CTRL : and all 3 only go to the rPI 40 pin header
 ** NO IMPACT ON EPAPER POWER SUPPLY **

 - 'scan_direction' is in fact PWR_EN, and switches 3V3 from the main LDO to the epaper power circuitry...
 ** is NOT turned off in epd_poweroff(), only in epd_poweroff_all() (which sets all the shift reg outputs to false.... **
** is set to true in epd_init() -> hence brownout as soon as driver is initialised, before even epd_poweron() is called! **

Where did you add the 20ms delay to avoid the brownout?

Is this right? If so, I think some fixes are required in the code:
- no need to do anything to 'power_disable', 'pos_power_enable' or 'neg_power_enable', they are not used on this board
-  scan_direction renamed to 'pwr_en', and set to false in epd_init(), and epd_poweroff()
** this should help low power operation as no longer driving the epaper boost circuit **

2. pwr_en impact on VBAT
- it seems like VBAT is switched directly onto the '5V' signal when pwr_en is true (by Q3). This seems like a bad thing to do to the battery when plugged in to the USB and hence there is really 5V on this trace (instead of charging it through the HX6610S 'normally').
- any explanation of what this circuit with Q3 is supposed to do?

thanks!

brianwyld, 2023-09-08:
Tried code as indicated above:
- saves me 1mA in lightsleep() mode
- saves about 30mA when not actively writing to the epaper
- peak inrush when turning on is about 400mA once running, 

attached some power captures (OTII on the battery connector @3.75V)
[image]
[image]
[image]

brianwyld, 2023-09-08:
One of my boards is very sensitive to the voltage drop, the other less so. A 470uF electrolytic across VDD_3V3 seems to fix it, but is kind of ugly.

brianwyld, 2023-09-12:
For a 'final' fix I went with a SMT 470uF tant capacitor as in the photo below : 
- the capa is the orange rectangle
- -ve side to GND from the SD card connector
-  +ve side wired directly to the ESP32 VDD_3V3 input pin 
- ignore the black wire which is left from previous tries with smaller MLCC caps)

[image]

random-0110-dude, 2024-08-03:
Hi all,

My board seems to have either a similar, or at least a related issue.

It works fine when I have battery attached to BAT. Without battery, it's very picky about power source. It works with some USB ports and non-brand USB wall adapters. It does not work with Apple 5W and 10W wall adapters, as well as with some power banks.

I was unable to find a correlation between voltage on VBUS and whether the board works. It works fine with some USB ports providing 4.85V and wall adapters with 5.1V, and does not work with other adapters that give 5.1V on VBUS.

What does it mean? When the board does not work, the green LED lights for a few seconds, and then red LED lights. The blue LED never starts.

This is what's happening on VBUS (4.85V):
[image]

In the same time, 3V3 line looks like (my oscilloscope is single-channel, so these are actually different measurements):
[image]
The voltage reaches 3.3V for a moment (green LED), and then drops to 1.04V (red LED). 

This is what's happening on VBAT:
[image]
The voltage reaches 4.18V for a moment (green LED), and then drops to 1.32V (red LED).

**Do you have any ideas what's going on here?** 

<hr>

I also tried the sketch in the first message here (with delays of 2000ms), this is what 3V3 line looks line on `epd_init();`

[image]

I haven't tried adding large capacitor to 3V3 yet. I also have other modifications: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/93#issuecomment-1503911075 (I removed the 22pF capacitors later, only 47uF is in place).

random-0110-dude, 2024-11-21:
Update: I added a 47uF electrolytic capacitor between GND and 3V3. Now the board works fine with all my power sources, except with one USB hub. Maybe I'll replace the capacitor with smth like 330/470 uF.

viteka32, 2024-11-21:
We are using less than a hundred LilyGo-EPD47s and we have noticed another problem that may be related to this behavior. Some PCBs have a very high voltage drop on the 750MA/0603 fuse (output after the fuse is about 1.6V) and only a jumper using a 0R resistor will help.

random-0110-dude, 2024-11-21:
@viteka32 could you please share a photo of your modifications?

viteka32, 2024-11-21:
I don't have a PCB now, but you can replace a fuse with a 0R resistor of size 0603.

[image]

random-0110-dude, 2024-11-21:
I checked the voltage between GND and fuse pads. On the bottom pad (as in the picture), it reads as +4.640V, while on the top pad it reads +3.585V. I guess my board is also affected?

pbenaim, 2024-11-21:
> For a 'final' fix I went with a SMT 470uF tant capacitor as in the photo below :
> 
> * the capa is the orange rectangle
> * -ve side to GND from the SD card connector
> * +ve side wired directly to the ESP32 VDD_3V3 input pin
> * ignore the black wire which is left from previous tries with smaller MLCC caps)
> 
> ![20230911_153007](https://private-user-images.githubusercontent.com/51411104/267319160-463c835b-761d-4067-ba44-157b220871fe.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIyMTU1MzMsIm5iZiI6MTczMjIxNTIzMywicGF0aCI6Ii81MTQxMTEwNC8yNjczMTkxNjAtNDYzYzgzNWItNzYxZC00MDY3LWJhNDQtMTU3YjIyMDg3MWZlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIxVDE4NTM1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM2NWY0YjdkNTIwNDlmZDU2MzNhM2JlM2FkMmE3NDc3MjVlNGRiZmRiNzUzYzczYmMzY2QxNWE2YWZlODQ2ZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Dmw_07dlYHM3gtZRv3T4oK5B1efHW2VQncEID4swl4M)

Yeeees the screen work (again) now :-)

But the SD Card and the Touch not working :-(

Demo Compiled with:
|-- SensorLib @ 0.1.9
|-- Button2 @ 2.3.2
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- WiFi @ 2.0.0

Any ideas ?

random-0110-dude, 2024-11-21:
@viteka32  btw my (V2.3) board has fuse with code D. Does it mean that it's 250 mA fuse, or I'm wrong?

@pbenaim One of the lib examples (in S3 branch) for SD card is wrong. Make sure your line says 
```SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);```

pbenaim, 2024-11-22:
> @pbenaim One of the lib examples (in S3 branch) for SD card is wrong. Make sure your line says `SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);`

Yes, but no way :-(

pbenaim, 2024-11-23:
So, for the moment, I use SPIFFS, for storage, but is not really cool 😎
For the actual project is enough, but maybe later I will need more space...

closed 2025-01-07 by github-actions[bot]