Xinyuan-LilyGO/LilyGo-EPD47 #100: Problem with function "epd_push_pixels(area, 50, 0);"  And the functions "epd_draw_image(area3, framebuffer, WHITE_ON_BLACK);"

SerjPr, 2023-08-25:
[image]

Hello. There are incomprehensible glitches. When using the epd_push_pixels(area, 50, 0); Should be a square. But on the sides there are additional lines. If the frame is in the middle, then there is only one line. When using the function
  epd_draw_image(area3, framebuffer, WHITE_ON_BLACK); A 50X50 square should be highlighted. But it turns out the line.

SerjPr, 2023-08-25:
#ifndef BOARD_HAS_PSRAM
#error "Please enable PSRAM !!!"
#endif
#include "epd_driver.h"
uint8_t *framebuffer = NULL;

Rect_t area = {
  .x = 700,
  .y = 50,
  .width = 200,
  .height = 200,
};
Rect_t area1 = {
  .x = 50,
  .y = 50,
  .width = 200,
  .height = 200,
};
Rect_t area2 = {
  .x = 400,
  .y = 50,
  .width = 200,
  .height = 200,
};
Rect_t area3 = {
  .x = 400,
  .y = 100,
  .width = 50,
  .height = 50,
};
void setup() {
  framebuffer = (uint8_t *)ps_calloc(sizeof(uint8_t), EPD_WIDTH * EPD_HEIGHT / 2);
  if (!framebuffer) {
    Serial.println("alloc memory failed !!!");
    while (1)
      ;
  }
  memset(framebuffer, 0xFF, EPD_WIDTH * EPD_HEIGHT / 2);
  epd_init();
  epd_poweron();
  delay(10);
  epd_clear();
  epd_push_pixels(area, 50, 0);
  epd_push_pixels(area1, 50, 0);
  epd_push_pixels(area2, 50, 0);
  epd_draw_image(area3, framebuffer, WHITE_ON_BLACK);
  epd_poweroff();
}
void loop() {
}

brianwyld, 2023-09-07:
My experience is that this problem is due to the byte/pixel  ordering in the driver. display update is line by line _in landscape orientation_ and alternate pixels are swapped.

Are you using a T5 (ESP32) or a T5-PLUS (ESP32S3), as the underlying IO transfer to the epaper uses a different driver (I2S on ESP32, esplcd on ESP32S3)?

SerjPr, 2023-09-08:
> My experience is that this problem is due to the byte/pixel ordering in the driver. display update is line by line _in landscape orientation_ and alternate pixels are swapped.
>
> Are you using a T5 [...]

I'm using ESP32-S3. I'm trying to do a partial screen refresh. There are examples of epd_clear_area(area); But this causes the update area to flicker. "epd_push_pixels" Causes two spurious streaks. And “epd_draw_image(area3, framebuffer, WHITE_ON_BLACK);” should highlight the specified area and highlights a horizontal stripe across the entire screen.

brianwyld, 2023-09-12:
I ended up doing a lot of refactoring in the driver to get something I understood  and that works for me with some issue still existing:
- partial refresh still degrades the untouched area after multiple updates
- faint lines round the updated areas 
- not working well on ESP32 (ie with the I2S DMA code - sometimes does a 'double image' ie right shift by about 10 pixels of some of the N frames - some timing/sync issue when writing the row out with the DMA? or the RMT?)

the code is so changed from the original methods I don't think I would be able to propose it as a PR (especially as I wasn't aiming to keep any backwards compatibily as I wanted to use it from micropython)... maybe I could put it up on my github as a 'inspired by'?

closed 2024-12-02 by github-actions[bot]