Xinyuan-LilyGO/LilyGo-EPD47 #101: compile error with demo sketch

HDMatGIT, 2023-08-26:
With the demo sketch I get this compilation error: 
C:\Users\hdm\AppData\Local\Temp\.arduinoIDE-unsaved2023726-3632-1dsa4p0.isof\demo\demo.ino:15:10: fatal error: pcf8563.h: No such file or directory
 #include "pcf8563.h"
          ^~~~~~~~~~~
compilation terminated.
Mehrere Bibliotheken wurden für "SD.h" gefunden
  Benutzt: C:\Users\hdm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SD
  Nicht benutzt: C:\Users\hdm\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: pcf8563.h: No such file or directory

I tried to install according libraries but none of them worked and produced further errors.
Can somebody point me to what is missing or the correct library?
Thanks

G6EJD, 2023-08-26:
You need to install the 8653 library there are plenty available.

HDMatGIT, 2023-08-26:
Thank you. I did that and tried some different ones. In none of them Line 42 of the demo sketch (PCF8563_Class rtc;) did work...,

G6EJD, 2023-08-26:
https://github.com/Bill2462/PCF8563-Arduino-Library

G6EJD, 2023-08-26:
Lily go should really have included links to all the libraries they use.

HDMatGIT, 2023-08-26:
Thank you again! I tried this library already yesterday, but then I get the error:
Compilation error: 'PCF8563_Class' does not name a type; did you mean 'PCF8563_H'?
I then tried PCF8563_H rtc; and then I get this error:
Compilation error: 'rtc' does not name a type

HDMatGIT, 2023-08-26:
I also tried "PCF8563 rtc;" This produces a new error:
Compilation error: 'class PCF8563' has no member named 'begin'

G6EJD, 2023-08-26:
Try the Adafruit rtclib

HDMatGIT, 2023-08-26:
That did not change anything.

lbuque, 2023-08-26:
https://github.com/lewisxhe/PCF8563_Library.git
The url of the library is written in platformio.ini.

HDMatGIT, 2023-08-26:
Thank you Ibuque that did the trick!

HDMatGIT, 2023-08-26:
Great and quick help, thank you!

closed 2023-08-26 by HDMatGIT