Xinyuan-LilyGO/LilyGo-EPD47 #102: T5 4.7 v2.3 bringup

malachib, 2023-10-23:
I'm attempting to use the `drawImages` example.
Compiling with IDF version 5.1.1
After forking and adding references to some extra soc gpio headers, I got it to compile and flash
However, when running, we see:

```
I (460) I80: Initialize Intel 8080 bus
E (470) esp_clk_tree: esp_clk_tree_src_get_freq_hz(21): unknown clk src
E (480) lcd_panel.io.i80: lcd_i80_select_periph_clock(495): get clock source frequency failed
E (480) lcd_panel.io.i80: esp_lcd_new_i80_bus(156): select periph clock 0 failed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42009061
```

Please advise, thank you

malachib, 2024-08-17:
It would appear ESP-IDF direct usage is not supported.

malachib, 2024-08-17:
Are there any ESP-IDF examples, instructions?  How about a datasheet for the EDO47TC1?

malachib, 2024-11-18:
Can't say I disagree with that assessment.

gmorell, 2024-12-13:
I came across this same set of issues with the esp-idf build  as well, I attempted to see if there were any changes [here](https://github.com/Xinyuan-LilyGO/lilygo-micropython/tree/5c6bc772971110dd9fdc3d7a05b2eb382319f45b/libs/LilyGo-EPD47/patches) that would be able to work with the library, but I was unable to get any changed behavior out of them.

voltagex, 2025-01-10:
> Are there any ESP-IDF examples, instructions?  How about a datasheet for the EDO47TC1?

is it this?

https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/esp32s3/datasheet/ED047TC1.pdf

malachib, 2025-01-10:
> > Are there any ESP-IDF examples, instructions?  How about a datasheet for the EDO47TC1?
> 
> is it this?
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/esp32s3/datasheet/ED047TC1.pdf

I saw that before and dismissed it since it was only 5 pages.  Now I realize there's actually 24 pages in there with a lot of detail.  Thanks for bringing that one up

malachib, 2025-02-10:
Allow me to un-stale it.  I have been working with a different EPD panel and plan to come back to this one

closed 2025-03-27 by github-actions[bot]

malachib, 2025-09-02:
@ogoidmatos I never had time to pick it up again.  I really would like to one of these days