Xinyuan-LilyGO/LilyGo-EPD47 #103: Compilation error: invalid preprocessing directive #i with font converted with fontconvert.py

sebyldino, 2023-10-26:
When i convert a .ttf font or .ttc with fontconvert.py

I get this error when compiling with Arduino IDE 2.2.1:

`#include "font/cambria.h"`

```
`D:\...\font\cambria.h:3:4: error: invalid preprocessing directive #i

  c o n s t   u i n t 8 _ t   c a m b r i a B i t m a p s [ 1 1 5 1 4 ]   =   {   
    ^

Compilation error: invalid preprocessing directive #i`
```

Font file .h created:
```
#pragma once
#include "epd_driver.h"
const uint8_t cambriaBitmaps[11514] = {...};
const GFXglyph cambriaGlyphs[] = {...};
const UnicodeInterval cambriaIntervals[] = {
    { 0x20, 0x7E, 0x0 },
};
const GFXfont cambria = {
    (uint8_t*)cambriaBitmaps,
    (GFXglyph*)cambriaGlyphs,
    (UnicodeInterval*)cambriaIntervals,
    1,
    1,
    44,
    36,
    -9,
};
```
Am I doing something wrong?

Thank you

sebyldino, 2023-10-26:
For example this font works:

`#include "font/opensans12.h"`

opensans12.h file:
```
#pragma once
#include "epd_driver.h"
const uint8_t OpenSans12Bitmaps[14575] = {...};
const GFXglyph OpenSans12Glyphs[] = {...};
const UnicodeInterval OpenSans12Intervals[] = {
    { 0x20, 0x7E, 0x0 },
    { 0xA0, 0xFF, 0x5F },
};
const GFXfont OpenSans12 = {
    (uint8_t*)OpenSans12Bitmaps,
    (GFXglyph*)OpenSans12Glyphs,
    (UnicodeInterval*)OpenSans12Intervals,
    2,
    1,
    34,
    27,
    -8,
};
```

gronpeter, 2024-02-04:
Hi sebyldino, I had the same problem, I am using windows 10. I had to convert the font file to UTF-8 after I have generated it. E.g. with notepad++. Now it works.

closed 2024-12-01 by github-actions[bot]