Xinyuan-LilyGO/LilyGo-EPD47 #106: Library error compiling sample code

NetCzech, 2023-12-24:
Hi,
I have LilyGo screen ED047TC1 (screen - 4,7 - S3, v2.3, 2021-6-10). I am trying to upload this simple code

```cpp
#include <Arduino.h>
#include "epd_driver.h"
#include "pic1.h"
#include "firasans.h"

void setup()
{
    Serial.begin(115200);

    // Initialization T5
    epd_init();

    Rect_t area1 = {
        .x = 30,
        .y = 100,
        .width = pic1_width,
        .height =  pic1_height
    };

    // Power on T5
    epd_poweron();

    // Clear T5
    epd_clear();

    // Draw logo
    epd_draw_grayscale_image(area1, (uint8_t *) pic1_data);
    delay(200);
    //epd_poweroff();

    //epd_poweron();
    int32_t cursor_x = 30;
    int32_t cursor_y = 320;
    char *string1 = "➸ LilyGo T5 4,7\"\n";
    
    // Draw first text
    writeln((GFXfont *)&FiraSans, string1, &cursor_x, &cursor_y, NULL);
    delay(200);
    cursor_x = 30;
    cursor_y += 50;
    char *string2 = "➸ 16 odstinu sedi\n";

    // Draw second text
    writeln((GFXfont *)&FiraSans, string2, &cursor_x, &cursor_y, NULL);
    delay(200);
    cursor_x = 30;
    cursor_y += 50;
    char *string3 = "➸ exkluzivne na laskarduino.cz";

    // Draw third text 
    writeln((GFXfont *)&FiraSans, string3, &cursor_x, &cursor_y, NULL);
    delay(200);

    // Power off T5
    epd_poweroff();
    
}

void loop()
{
 
}
```

I always get a driver error during compilation

```
In file included from d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:6:
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:114:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_output_row'
  114 | void IRAM_ATTR epd_output_row(uint32_t output_time_dus);
      |                ^~~~~~~~~~~~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:119:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_skip'
  119 | void IRAM_ATTR epd_skip();
      |                ^~~~~~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:124:21: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_get_current_buffer'
  124 | uint8_t * IRAM_ATTR epd_get_current_buffer();
      |                     ^~~~~~~~~~~~~~~~~~~~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:132:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_switch_buffer'
  132 | void IRAM_ATTR epd_switch_buffer();
      |                ^~~~~~~~~~~~~~~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c: In function 'fast_gpio_set_hi':
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:58:5: error: 'GPIO' undeclared (first use in this function)
   58 |     GPIO.out_w1ts = (1 << gpio_num);
      |     ^~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:58:5: note: each undeclared identifier is reported only once for each function it appears in
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c: In function 'fast_gpio_set_lo':
d:\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:63:5: error: 'GPIO' undeclared (first use in this function)
   63 |     GPIO.out_w1tc = (1 << gpio_num);
      |     ^~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c: In function 'gpio_setup_out':
d:\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:21: error: 'GPIO_PIN_MUX_REG' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                     ^~~~~~~~~~~~~~~~
d:\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:21: note: each undeclared identifier is reported only once for each function it appears in
d:\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:45: error: 'PIN_FUNC_GPIO' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                                             ^~~~~~~~~~~~~
```

I could ask you for help on how to solve this problem.
Thank you

lewisxhe, 2024-01-03:
Reduce arduino esp32 to below V3.0

NetCzech, 2024-01-07:
> Reduce arduino esp32 to below V3.0

It's work,
thank you

closed 2024-01-07 by NetCzech