Xinyuan-LilyGO/LilyGo-EPD47 #114: Partial refresh - irreversible damage?

mirko3000, 2024-03-14:
Hi all,

on the product page of the T5 ePaper I read the following statement:
"Due to the problem of the characteristics of the ink screen, it cannot be partially refreshed for a long time, otherwise, it will leave residual images and lead to irreversible damage."

Can someone explain me to what extend partial refresh can be used? Is it enough to do a full refresh every once in a while to avoid kind of "burn in" of static content? 
I would like to reduce the "flickering" of the screen on every update, thinking of doing partial refresh and a full refresh e.g. every hour.

Best regards,
Mirko

mirko3000, 2024-03-15:
For now I am using this workaround to reduce flickering:

`epd_clear_area_cycles(epd_full_screen(), 1, 50);`

This does only a single refresh and still gives me good results. Every 30 min I do the full regular refresh with 4 cycles.

MrMutlu, 2024-05-10:
Hi, i ordered an e-paper and saw cauton on vendor page. If you used that for a while can you share your experience for avoiding "burn in" or "unrepairable damage" issues? Many thanks.

closed 2024-11-17 by github-actions[bot]